confidential-containers/td-shim #1: Consider renaming the repo

sameo, 2021-11-12:
Although this repo is TDX specific for now, it may be able to support other TEEs. If there is both interest for that and technical feasibility, we should consider renaming it to e.g. `cc-shim`.

jyao1, 2021-11-12:
Good suggestion. Agree.

jiangliu, 2021-11-12:
> Although this repo is TDX specific for now, it may be able to support other TEEs. If there is both interest for that and technical feasibility, we should consider renaming it to e.g. `cc-shim`.

A repo with prefix "cc" will be great, that may be first crate designed for cc:)

c3d, 2021-11-18:
If you rename, there is also a bit of renaming in the design documents and other places, since there are many `td-` prefixes there as well.

jodh-intel, 2021-11-18:
`cc-shim` makes sense as most of us have probably forgotten about [this `cc-shim`](https://github.com/clearcontainers/shim) by now :smile:

jiangliu, 2021-11-22:
> `cc-shim` makes sense as most of us have probably forgotten about [this `cc-shim`](https://github.com/clearcontainers/shim) by now 😄

that's a long long story😄

jodh-intel, 2021-11-22:
@jiangliu - 哈哈! Yeah :smile:

ariel-adam, 2022-10-11:
@sameo dude, is this issue still relevant or can be closed? 
If it's still relevant to what release do you think we should map it to (mid-November, end-December, mid-February etc...)?