confidential-containers/td-shim #3: Switch to release versions of Rust build tools when possible

c3d, 2021-11-30:
Right now, the stable version of the Rust toolchain is not sufficient to build td-shim.

This issue is to acknowledge that this is known and will be fixed when the stable tool chain is good enough.

See for example comments by @jodh-intel and @c3d, and response by @jyaho1 in https://github.com/confidential-containers/td-shim/pull/2#r752365817_

jiangliu, 2022-01-23:
@jyao1 any details information about why stick to nightly toolchain? We are trying to avoid using nightly toolchain in product:)

jiangliu, 2022-01-24:
I'm trying to figure out a way to use rust stable toolchain. It may help to use dev container for tdshim. Does any test depend on TDX hardware or kvm? If no, feel it's feasible to create a container with dev environment for tdshim. 
@jyao1 @sameo

xiaoyuxlu, 2022-01-24:
Hi jiangliu, I'm sorry that rust stable toolchain doesn't support in short-term. Currently some unstable features like ```#![feature(global_asm)]``` ```#![feature(alloc_error_handler)]``` etc is used. That require nightly rust.

> Does any test depend on TDX hardware or kvm?  
No. 
> If no, feel it's feasible to create a container with dev environment for tdshim.
I think it's a good suggestion. This makes it easy to set up a development environment. But if you want to launch a TD you must have TDX hardware and TDX software stack.

jyao1, 2022-01-24:
Today, most firmware, bootloader, non-application projects are using nightly, for example:
- https://github.com/rust-osdev/bootloader
- https://github.com/rust-osdev/uefi-rs
- https://github.com/cloud-hypervisor/rust-hypervisor-firmware
- https://github.com/oreboot/oreboot
- https://github.com/tock/tock
- https://github.com/rustsbi/rustsbi

I think we have to wait until stable has better support for no_std.

xiaoyuxlu, 2022-01-24:
Another reason for using nightly is that current target ```x86_64-unknown-uefi``` is under "Tier 3" see: https://doc.rust-lang.org/nightly/rustc/platform-support.html. This means there is no official pre-build binary release. thus, we can't use ```rustup target add x86_64-unknown-uefi``` to add target libraries. So, we must cross build the libraries for ```core``` ```alloc``` libraries.  Rust provide a method called [build-std](https://github.com/rust-lang/wg-cargo-std-aware#mvp-implementation) for that. It also need a nightly rust-toolchain.

jiangliu, 2022-01-24:
Rust toolchain always depends nightly version to bootstrap. Once we have the sysroot (core) ready, we could build all our own crates with stable toolchains. Traditionally xargo/xbuild is used, which build the rust no_std library and all our own crates with nightly toolchain. Recently there's a new crate `sysroot`, https://github.com/DianaNites/cargo-sysroot, which build the no_std library with nightly toolchain and build our own crates with stable toolchain. 
I think we may follow sysroot if possible.

xiaoyuxlu, 2022-01-25:
> Recently there's a new crate sysroot, https://github.com/DianaNites/cargo-sysroot, which build the no_std library with nightly toolchain and build our own crates with stable toolchain.

Sounds good! Have you tried it?

jiangliu, 2022-01-25:
Working on it, just resolved a bug of cargo-sysroot:
https://github.com/DianaNites/cargo-sysroot/issues/21#issuecomment-1020591314

ariel-adam, 2022-10-11:
@c3d is this issue still relevant or can be closed? 
If it's still relevant to what release do you think we should map it to (mid-November, end-December, mid-February etc...)?