confidential-containers/td-shim #4: Make the global allocator thread safe

c3d, 2021-11-30:
I have issues with memory ordering in the allocator as written.

For some reason, I updated a thread that was marked as "resolved", and it did not "unresolve" it. So here is [a link to the comment in context](https://github.com/confidential-containers/td-shim/pull/2/files/3eb0cbc246a5eeb72e4747a978920fe6790b6786#r759165742), and here is a copy of that comment, because at least for me, the link to the resolved comment does not open it 😠 :

### Original comment (slightly edited for clarity out of context)

The comment about the `GlobalAlloc` trait being `unsafe` is useful. Please add a link to this: https://doc.rust-lang.org/std/alloc/trait.GlobalAlloc.html.

But I'm sorry, I was actually referring to the unsafe wrapper and const-casting around `used_heap` and `max_heap` updates inside the implementation. In my opinion, the fact that the _interface_ requires an `unsafe impl` does not mean that it's a good idea to add unsafe blocks inside (quite to the contrary, actually, you need to be double plus careful).

For instance, the original example in the URL above has things like:

```
        if self
            .remaining
            .fetch_update(SeqCst, SeqCst, |mut remaining| {
                if size > remaining {
                    return None;
                }
                remaining -= size;
                remaining &= align_mask_to_round_down;
                allocated = remaining;
                Some(remaining)
            })
            .is_err()
```

That sample implementation takes full advantage of the Rust type system, and is actually somewhat safer than yours. As you can see, it does not actually use any `unsafe` _block_ (what I referred to), only `unsafe` _interfaces_.

Here are a few things I see as at least potentially problematic in your code:

* It is not thread safe, because it uses regular additions (with no obvious locking) instead of atomic operations. The sample code does use `fetch_update`, which is atomic.
* It is also not thread safe because it splits tests and the consequence of the tests (this is related to the above). For example, the `if self.max_heap < self.used_heap` condition could be true on thread 1, then thread 2 does an allocation, updates max_heap to some larger value than what you have, and by the time you do your assignment below, you overwrite the `max_heap` field with a bogus value. See below details of the sequence.

In short, either you tell me that this code is strictly single-threaded, or that it only deals with allocators that cannot possibly be shared between two threads, and then you have to put a fat large comment explaining that. Or your code is wrong as written. I don't know which one it is 😉.

Here is a possible (and likely) sequence of events for the code as written (illustrated with the second `if` in `alloc`, but alloc has at least two instances of this problem and dealloc at least one, probably two. Let's assume two threads A and B.

| Thread A | Thread B | used_heap | max_heap | Comment |
| :--------: | :--------: | ----------: | ----------: | ---------- |
| alloc(13)  | alloc(17)  | 0 | 0 | Two threads enter alloc |
| x = 13 | x = 17 | 0 | 0 |  Each thread reads `layout.size()` |
| y = 0  | y = 0 | 0 | 0 | A reads `used_heap`, so does B |
| z = 13 | z = 17 | 0 | 0 | A and B both compute new `used_heap` value |
| write z | write z | 17 | 0 | A and B both write the value. B writes last, so `used_heap` is now `17`. |
| t = 0 | t = 0 | 17 | 0 | A and B both read `max_heap` to be `0` |
| `if t < z` checking `0 < 13` | `if t < z` checking `0 < 17` | 17 | 0 | **OOOPS** A may still use some cached / optimized value of `used_heap`, since unsafe blocks or functions have no impact on memory ordering semantics AFAICT. |
| A writes 13 in `max_heap` | B writes 17 in `max_heap` | 17 | 13 | **OOOPS** You were unlucky, this time A happens to write last, so now you have an inconsistent internal state |

So what I need is:

* Either a comment explaining why the above was carefully considered and cannot possibly happen, for example because you know your code only works on the boot CPU and there is no AP yet.
* Or a rewrite of that code to [use atomics properly](https://riptutorial.com/rust/example/21259/atomics-and-memory-ordering)

_Originally posted by @c3d in https://github.com/confidential-containers/td-shim/pull/2#pullrequestreview-818872111_

jyao1, 2021-12-09:
The global allocator in benchmark is for test purpose only. It is not thread safe by design. We only allow BSP use heap.
See https://github.com/confidential-containers/td-shim/blob/staging/benchmark/src/lib.rs#L56

c3d, 2021-12-10:
> The global allocator in benchmark is for test purpose only.

Shouldn't it be under `#[cfg(test)` or `#[cfg(feature="benches")]` in that case?

Also, what kind of tests? Asking because "benchmark" seems to suggests that we are talking about performance testing. In that case, won't you introduce a bias by having the global allocator be "over-optimized" ?

>It is not thread safe by design.

Unsafe by design? Hmm…

> We only allow BSP use heap.

This is not enforced anywhere that I can see (except now a comment after I asked for it, thanks). And frankly, it's not hard to make it correct. So please, let's consider this a bug and not a feature 😄 (and not an urgent one to fix at that, don't worry).

> See https://github.com/confidential-containers/td-shim/blob/staging/benchmark/src/lib.rs#L56

ariel-adam, 2022-10-11:
@c3d is this issue still relevant or can be closed? 
If it's still relevant to what release do you think we should map it to (mid-November, end-December, mid-February etc...)?

c3d, 2022-10-12:
@ariel-adam The bad code is still there.

ariel-adam, 2022-10-12:
Ack, I've put it on the CoCo releases project with not target drop.
Lets see in the future if someone jumps on it and we close it at some point