confidential-containers/td-shim #8: Inconsistent interfaces compared to TDVF

sboeuf, 2021-12-14:
TD Shim is described as a lightweight version of TDVF. In theory that should only mean it supports less features compared to what's supported by TDVF, but in practice we can see the interfaces are also different and the expectations on what can be passed from the VMM are different.
I think the goal is to be able to run both TDVF and TD Shim with the same internal implementation from the VMM, meaning that we can rely on the same interfaces.

__Direct kernel boot__

If we decide to support direct kernel boot by allowing `Payload` and `PayloadParam` as part of the TDVF section list for TD Shim, then the same mechanism should be supported by TDVF as well.

__ACPI tables__

TDVF supports any kind of ACPI table (both static and dynamic), but TD Shim only supports static ones, meaning the DSDT and SSDT can't be provided by the VMM.

@rbradford mentioned:
>Not including the DSDT is, from my understanding, motivated by not wanting to support hotplug. However the DSDT serves multiple purposes in Cloud Hypervisor beyond hotplug:
>
>    PCI routing information
>    Serial port IRQ setup
>    Multiple PCI segment support
>    Specifying the value for the ACPI HW reduced shutdown register to trigger a shutdown
>    Power button triggering through GED
>
>Further I think that it is possible to support PCI device hotplug with TDX.
>
>The DSDT is specified as part of the FADT/FACP so the kernel will find it anyway if you give it the FADT table unless the TD shim to >manually create it's own RSDP and FADT tables.
>
>A much simpler approach would be to just specify the RSDP address in the HOB (as per the TDVF based boot.)

We need some consistency on how ACPI tables are shared with TD Shim and TDVF, the address to the RSDP being the simplest one.

And the fact that TD Shim (because it doesn't support DSDT) requires a new table to be introduced in the ACPI spec (SPRT) feels like too much. If the requirement around not supporting dynamic table is about security, then either the VMM shouldn't expose a DSDT or the guest kernel shouldn't include the ASL/AML compiler. The requirement should not be part of the firmware itself.

sboeuf, 2021-12-14:
/cc @jyao1 @jiangliu

jyao1, 2021-12-14:
I don't like the idea to pass RSDP address. The reason is 1) all guest memory shall be managed by TD-Shim, 2) all VMM input shall be extended.

If I understand correctly, you just want to have a way to let VMM pass DSDT and TD-Shim put DSDT to final FADT, right?

sboeuf, 2021-12-14:
>I don't like the idea to pass RSDP address. The reason is 1) all guest memory shall be managed by TD-Shim, 2) all VMM input shall be extended.

I'm fine with that, but all we're looking for is consistency between the two versions of the firmware. As a reminder, for TDVF to support Cloud Hypervisor, we agreed on a solution that relies on passing the entire set of ACPI tables at once by copying them to the HOB. Adopting the same mechanism for TD Shim (given that TD Shim would accept all kinds of tables) would be a nice way not having to define a new mechanism dedicated to TD Shim.

>If I understand correctly, you just want to have a way to let VMM pass DSDT and TD-Shim put DSDT to final FADT, right?

Yes, there are important information provided by the VMM in the DSDT that are worth being passed to the guest. And there's no need/reason for the firmware to prevent such information from being communicated to the guest OS.

jyao1, 2021-12-15:
> As a reminder, for TDVF to support Cloud Hypervisor, we agreed on a solution that relies on passing

I disagree. We never agree on a solution like this. Any misunderstanding?

sboeuf, 2021-12-15:
> > As a reminder, for TDVF to support Cloud Hypervisor, we agreed on a solution that relies on passing
> 
> I disagree. We never agree on a solution to pass RSDP address. Any misunderstanding?
> 
> I think all ACPI tables information should be in HOB, but not only RSDP address. The TDVF/TD-Shim should get the table one by one and reconstruct the RSDP/XSDT/...

We agreed on a solution to pass the RSDP address along with the whole data size, given that we were copying the data (in that case the entire set of ACPI tables) to the HOB.

But I'm fine with updating this solution if you think passing each table one by one is a better alternative, and as long as both TDVF and TD Shim rely on the same mechanism.

jyao1, 2021-12-15:
> We agreed on a solution to pass the RSDP address

Let's get clear who is "we". I am sure I am not included there.

jyao1, 2021-12-15:
> as long as both TDVF and TD Shim rely on the same mechanism.

Yes. that is our goal. Let's move to this direction.

sboeuf, 2021-12-15:
> > as long as both TDVF and TD Shim rely on the same mechanism.
> 
> Yes. that is our goal. Let's move to this direction.

That's great!

About the need for a DSDT, would you be okay adding the support for it in TD Shim so that it can be passed all the way from the VMM into the guest?
And that way, both firmware versions would support it.

jyao1, 2021-12-15:
One possible option I was thinking is that:
```
If VMM passes DSDT in ACPI Hob, then the Td-Shim links DSDT to FADT.DSDT.   <== New step
Else, Td-Shim fill zero to FADT.DSDT field.
```
Then we can support both cases - with or without DSDT.

@jiangliu, what do you think?

rbradford, 2021-12-15:
When patching the tables I think it would be good to make sure to update the creator fields on the SDT to make it clear that the TD shim has modified the table. This would necessitate generating a new checksum but that will also be necessary for tables such as the FADT/FACP that point to the DSDT or the XSDP and RSDP which also point to the table addresses.

sboeuf, 2021-12-17:
ping @jiangliu

sboeuf, 2021-12-17:
Hi @jyao1, I have a question about TD Payload related to what the specification states:
```
If the VMM does not provide the payload image or the VMM does not know
the payload image format, the VMM shall not create this payload info HOB.
The TdShim must know the payload format and transfer to the payload in
right way. By default, PayloadImageTypeExecutablePayload is used.
```
Assuming the TD Shim has the ability to identify the payload type, why don't we always rely on the TD Shim to do so? I don't think we should expect the VMM to identify the payload type, especially since it's not doing anything with it other than passing it to the guest.

jyao1, 2021-12-17:
@sboeuf, can we move different question to different issue ? I think this issue will only discuss ACPI.

sboeuf, 2021-12-17:
Sounds good!

jiangliu, 2021-12-20:
> One possible option I was thinking is that:
> 
> ```
> If VMM passes DSDT in ACPI Hob, then the Td-Shim links DSDT table to FADT.DSDT field.   <== New step
> Else, Td-Shim fills zero to FADT.DSDT field.
> ```
> 
> Then we can support both cases - with or without DSDT.
> 
> @jiangliu, what do you think?

I think it's ok to optionally support DSDT, then the vmm/client could make the decision.

closed 2021-12-25 by jyao1