confidential-containers/td-shim #9: Payload type detection

sboeuf, 2021-12-17:
I have a question about TD Payload related to what the specification states:
```
If the VMM does not provide the payload image or the VMM does not know
the payload image format, the VMM shall not create this payload info HOB.
The TdShim must know the payload format and transfer to the payload in
right way. By default, PayloadImageTypeExecutablePayload is used.
```
Assuming the TD Shim has the ability to identify the payload type, why don't we always rely on the TD Shim to do so? I don't think we should expect the VMM to identify the payload type, especially since it's not doing anything with it other than passing it to the guest.

jyao1, 2021-12-17:
You are right. If TD shim has ability to identity, then there is no need to let VMM pass the info.

For example, the VMM does not tell TD shim the detail of PayloadImageTypeExecutablePayload (PE/COFF image, or ELF image), the TD shim can detect it is PE/COFF or ELF.

But if TD shim cannot detect, then TD shim relies on VMM to pass the info.

What is your concern? or your proposal?

sboeuf, 2021-12-17:
Well my understanding from the spec is that TD Shim will __always__ be able to detect the payload type. I'm suggesting that if that's the case, then there's no point in having `HOB_PAYLOAD_INFO_TABLE` at all. The TD Shim should already know where the Payload is located since it provides the information through the TDVF sections already.
Basically I'm proposing some simplification of the spec, by removing the need for the VMM to provide `HOB_PAYLOAD_INFO_TABLE`.

jyao1, 2021-12-20:
> TD Shim will **always** be able to detect the payload type.

I doubt if this is TRUE.
How TD shim know if a binary blob is PayloadImageTypeBzImage or PayloadImageTypeRawVmLinux ?

sboeuf, 2021-12-21:
> > TD Shim will **always** be able to detect the payload type.
> 
> I doubt if this is TRUE. 

Oh okay then I guess I read the spec the wrong way. Maybe something like the following would bring some clarity:
```
The VMM must always provide the image format to the TD Shim through
the HOB because the TD Shim might not be able to detect the payload
format.
```

> How does TD shim know if a binary blob is PayloadImageTypeBzImage or PayloadImageTypeRawVmLinux ?

Well the same way you're suggesting the VMM will need to have some detection code, the TD Shim could have the same kind of detection mechanism.

But more than that, since the VMM is not the one preparing the payload and the boot parameters, it could be completely opaque and simply let the TD Shim do the detection.

jyao1, 2021-12-21:
Sorry for the confusing. I would say in this way:

1. If TD shim includes payload and knows the payload type, then TD shim does not rely on VMM input. TD shim can boot payload directly.

2. If TD shim does not include payload, and rely on VMM input the payload, then VMM knows the payload and provides the payload type.

sboeuf, 2021-12-21:
Oh I see the payload can be included in the TD Shim binary. I didn't get that, and that explains why you expect the TD Shim to know the type in that case.

And just to clarify, is there any payload type detection code in TD Shim? Some code that looks at the payload and understands the type?

jyao1, 2021-12-22:
TD Shim may *know* or *assume* Payload Type at build time. But I don't think TD Shim can *detect* Payload Type at runtime.
I will add some clarification in the spec.

closed 2021-12-25 by jyao1

sboeuf, 2022-01-04:
Hi @jyao1, based on @rbradford's input, here are two main reasons why the TDShim should be able to detect the payload type:
1. New support for different payload types can't be added unless every VMM is modified
2. The VMM could "lie" to the firmware which could push it down a path that means it could then be potentially exploitable (since the firmware runs inside the TD)

I wanted to have your thoughts on this.

jyao1, 2022-01-05:
> * New support for different payload types can't be added unless every VMM is modified
VMM is already modified if VMM wants to support and provide new PAYLOAD in metadata section.
VMM is not required to modify (provide payload type hob), if VMM does not support new PAYLOAD in metadata section.
I don't understand why there is concern.

> * The VMM could "lie" to the firmware which could push it down a path that means it could then be potentially exploitable (since the firmware runs inside the TD)
In TDX, VMM already can lie almost everything to TD.
Before attestation, you cannot trust a TD, because VMM may provide a malicious TD. 
In order to ensure you are running the environment you want to boot, you have to use attestation.
I don't understand why there is concern.

sboeuf, 2022-01-05:
> > * New support for different payload types can't be added unless every VMM is modified
> 
> VMM is already modified if VMM wants to support and provide new PAYLOAD in metadata section. VMM is not required to modify (to provide payload type hob), if VMM does not support new PAYLOAD in metadata section. I don't understand why there is concern.

The point is that if the firmware is able to detect the type, we wouldn't need to define a new GUID type and a list of accepted payload types, meaning there would be no changes required from the VMM's perspective.

> 
> @sboeuf, If you think tdshim need detect, then please share your algorithm on how to detect those 3 different types of payload.
> 

Well to differentiate between vmlinux and bzImage, you can check if you find the magic numbers to identify an ELF, and if not present, you could check if this is a bzImage by checking the header magic number at offset `0x202`. And in case you don't think it's either a vmlinux or a bzImage, you fallback onto the last possibility.

> In TDX, VMM already can lie almost everything to TD. Before attestation, you cannot trust a TD, because VMM may provide a malicious TD. In order to ensure you are running the environment you want to [...]

Fair enough :)

jyao1, 2022-01-05:
> Well to differentiate between vmlinux and bzImage, you can check if you find the magic numbers to identify an ELF

This is wrong. The ELF header is stripped for PayloadImageTypeRawVmLinux.

sboeuf, 2022-01-05:
> > Well to differentiate between vmlinux and bzImage, you can check if you find the magic numbers to identify an ELF
> 
> This is wrong. The ELF header is stripped for PayloadImageTypeRawVmLinux.

Oh okay, but then I don't understand why it works for Cloud Hypervisor to rely on https://github.com/rust-vmm/linux-loader/blob/main/src/loader/x86_64/elf/mod.rs#L137-L156 to load `vmlinux` kernel?

sboeuf, 2022-01-05:
We really don't want to have the detection code to be added to Cloud Hypervisor when TDShim could be the one doing it.
Otherwise this would duplicate the detection code between the VMM and the firmware.
And we don't think it's appropriate to rely on the user to provide that information when there's a way to detect it.

I can send a PR to TDShim for the detection if you're okay with it.

jyao1, 2022-01-06:
Anyone can submit PR. We can review what you propose.

Please remember we need support 3 types of payload defined in the current TdShim spec.