confidential-containers/td-shim #34: Build td-shim/ staging fails

jiangliu, 2022-01-24:
root@a484e09e9ade:/tdshim# cargo xbuild -p rust-tdshim --target x86_64-unknown-uefi --release
WARNING: There is no root package to read the cargo-xbuild config from.
   Compiling ring v0.16.20 (https://github.com/jyao1/ring.git?branch=uefi_support#3e042088)
   Compiling scroll_derive v0.10.5
error: failed to run custom build command for `ring v0.16.20 (https://github.com/jyao1/ring.git?branch=uefi_support#3e042088)`

Caused by:
  process didn't exit successfully: `/tdshim/target/release/build/ring-71a085a804d92895/build-script-build` (exit status: 101)
  --- stdout
  OPT_LEVEL = Some("3")
  TARGET = Some("x86_64-unknown-uefi")
  HOST = Some("x86_64-unknown-linux-gnu")
  CC_x86_64-unknown-uefi = None
  CC_x86_64_unknown_uefi = None
  TARGET_CC = None
  CC = Some("clang")
  CFLAGS_x86_64-unknown-uefi = None
  CFLAGS_x86_64_unknown_uefi = None
  TARGET_CFLAGS = None
  CFLAGS = None
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some("false")

  --- stderr
  running "clang" "-O3" "-ffunction-sections" "-fdata-sections" "--target=x86_64-unknown-windows-gnu" "-I" "include" "-Wall" "-Wextra" "-std=c1x" "-Wbad-function-cast" "-Wnested-externs" "-Wstrict-prototypes" "-pedantic" "-pedantic-errors" "-Wall" "-Wextra" "-Wcast-align" "-Wcast-qual" "-Wconversion" "-Wenum-compare" "-Wfloat-equal" "-Wformat=2" "-Winline" "-Winvalid-pch" "-Wmissing-field-initializers" "-Wmissing-include-dirs" "-Wredundant-decls" "-Wshadow" "-Wsign-compare" "-Wsign-conversion" "-Wundef" "-Wuninitialized" "-Wwrite-strings" "-fno-strict-aliasing" "-fvisibility=hidden" "-g3" "-nostdlibinc" "-ffreestanding" "-Werror" "-DNDEBUG" "-DGFp_NOSTDLIBINC=1" "-c" "-o/tdshim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/aes_nohw.o" "crypto/fipsmodule/aes/aes_nohw.c"
  thread 'main' panicked at 'failed to execute ["clang" "-O3" "-ffunction-sections" "-fdata-sections" "--target=x86_64-unknown-windows-gnu" "-I" "include" "-Wall" "-Wextra" "-std=c1x" "-Wbad-function-cast" "-Wnested-externs" "-Wstrict-prototypes" "-pedantic" "-pedantic-errors" "-Wall" "-Wextra" "-Wcast-align" "-Wcast-qual" "-Wconversion" "-Wenum-compare" "-Wfloat-equal" "-Wformat=2" "-Winline" "-Winvalid-pch" "-Wmissing-field-initializers" "-Wmissing-include-dirs" "-Wredundant-decls" "-Wshadow" "-Wsign-compare" "-Wsign-conversion" "-Wundef" "-Wuninitialized" "-Wwrite-strings" "-fno-strict-aliasing" "-fvisibility=hidden" "-g3" "-nostdlibinc" "-ffreestanding" "-Werror" "-DNDEBUG" "-DGFp_NOSTDLIBINC=1" "-c" "-o/tdshim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/aes_nohw.o" "crypto/fipsmodule/aes/aes_nohw.c"]: No such file or directory (os error 2)', /root/.cargo/git/checkouts/ring-1836a5bf48f90a2a/3e04208/build.rs:660:9
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
^C  Building [=================>         ] 62/93: scroll_derive

gaojiaqi7, 2022-01-24:
hi @jiangliu it seems that **clang** cannot be found, can you please check if it is correctly configured on your host?

jiangliu, 2022-01-24:
Thanks, seems forgot to install clang.

closed 2022-01-24 by jiangliu