confidential-containers/td-shim #42: Status track for merging into main branch

jiangliu, 2022-01-28:
As the td-shim has been merged into the staging branch, we need a plan to eventually merge it into main branch. Let's get it done piece by piece. So a checklist table is maintained here to track the status.

Approvers may review and approve the project by components, and mark a component as ready for merging in the following table. One all components has received enough approves, the staging branch will be merged into the main branch. Anyone who is willing to help the migration process may add himself/herself into the below table as approvers.

| Approver | @c3d  | @jiangliu | @jyao1  | @sameo | 
| --- | --- | --- | --- | --- |
| Build System |  |  |
| rust-td-layout | |  |

c3d, 2022-01-28:
@jiangliu Thanks a lot for doing that. What about documentation / slides? Those are easier to merge first, no?

jiangliu, 2022-02-03:
> @jiangliu Thanks a lot for doing that. What about documentation / slides? Those are easier to merge first, no?

Yes, let's move along that way.
It would be great to eventually merge the staging branch into the main branch, so we could keep the development history.

jyao1, 2022-06-12:
staging branch is merged to main. Close this one.

closed 2022-06-12 by jyao1