confidential-containers/td-shim #52: Add unit cases for td-layout/memslice.rs

jiangliu, 2022-02-02:
Current there's no unit test cases for td-layout/memslice.rs, it would be better to add unit test cases for it.

liuw1, 2022-02-22:
I will take this task.

jodh-intel, 2022-02-22:
Thanks @liuw1!

@jyao1 and @liuw1 - Could the public functions be changed to return a `Result` instead of just calling `panic()`?

jiangliu, 2022-02-22:
I have tried to do that, and have converted some unwrap()/expected()/panic!() to Result. But there are still plenty callsites:(
On the other hand, it's firmware instead of normal application, panic with clear messages is acceptable too:)

jyao1, 2022-02-22:
Firmware is special.

The problem in firmware is that, even you can return result, you have to handle the situation to stop execution at some point. You cannot return back to hypervisor in case of critical error.

In general,
1) If we can continue and move forward boot into OS, then we should use Result.
2) If we cannot continue and have to stop booting, then using panic() is much better. At least you know what is wrong immediate at this point.

If module A gets some critical error, we should stop at module A, instead of stopping at a faraway module D.

PR #117 (2022-02-24): Add unit cases for td-layout/memslice.rs
Check some panic scenarios.

Fix #52
PR #119 (2022-02-24): Add unit test case for td-layout/memslice.rs
Check functionality and some panic scenarios.
Code is same with the PR for staging. This PR is for master.

Fix #52
closed 2022-02-24 by jyao1