confidential-containers/td-shim #81: Implement shim loader in td-shim-tools

jiangliu, 2022-02-10:
We have a linker `td-shim-ld` in the td-shim-tools crate, there should be a loader too.
The loader should take the responsibilities:
1) parse td-shim binary file into sections
2) validate section information
3) enforce present sections follow the td-shim spec

jyao1, 2022-02-14:
1) Create a `td-shim-loader` library, that could be reused by the VMM
2) Create a `td-shim-checker` tool (consuming `td-shim-loader`), to validate the format of the final binary.

jiangliu, 2022-02-15:
> 1. Create a `td-shim-loader` library, that could be reused by the VMM
> 2. Create a `td-shim-checker` tool (consuming `td-shim-loader`), to validate the format of the final binary.

Yes, exactly.
It may be included into the td-shim-tools crate.

mxu9, 2022-03-15:
As discussed with Jiewen, I will take this task and plan to start this work after I finish test-acpi-payload. ETA is ww15.

mxu9, 2022-03-29:
> parse td-shim binary file into sections
td-shim is of PE format. Here section means the PE section?

jyao1, 2022-03-30:
It is meta section, not PE section.

jyao1, 2022-05-16:
done

closed 2022-05-16 by jyao1