confidential-containers/td-shim #98: Does TD-Shim support PVH-style direct kernel boot?

dimakuv, 2022-02-16:
I couldn't find any information in this repo whether TD-Shim supports (or has plans to support) the PVH-style direct kernel boot, in addition to the traditional Linux Boot protocol.

For reference, PVH ABI is described here: https://xenbits.xen.org/docs/unstable/misc/pvh.html.

From what I understand, TD-Shim currently only supports the Linux Boot protocol (filling in the "zero page").

On the other hand, CloudHypervisor recently removed support for the Linux Boot protocol in favor of the PVH-style kernel boot: https://github.com/cloud-hypervisor/cloud-hypervisor/blob/95ca79974a32f3dab5987032361d0ecd2aa65512/release-notes.md#removed-functionality

At the same time, CloudHypervisor advertises its usage with TD-Shim: https://github.com/cloud-hypervisor/cloud-hypervisor/blob/072114d7c51fc42fd2a7b78a879018f13c8052e4/docs/intel_tdx.md#tdshim

This is confusing: how can TD-Shim be started with CloudHypervisor if TD-Shim *doesn't support* PVH and CloudHypervisor *only supports* PVH? Am I missing something?

jiangliu, 2022-02-16:
FYI: https://github.com/confidential-containers/td-shim/pull/79

sboeuf, 2022-02-16:
@dimakuv I understand the confusion but it does work.

The reason is, in case of Cloud Hypervisor performing direct kernel boot in the non-TDX case, this is Cloud Hypervisor responsibility to handle the protocol and to prepare what's needed before booting. And for this case, Cloud Hypervisor only supports a PVH kernel binary to be provided as it only implements this protocol.

In case of TDX, the firmware (TDShim) is the one implementing the protocol. That means even if Cloud Hypervisor doesn't support Linux Boot Protocol, it is simply copying the kernel somewhere in memory so that the TDShim can handle the filling of zero page and the kernel boot parameters.

Hope this helps clarifying.

jyao1, 2022-02-16:
thanks @sboeuf to clarify that.

@dimakuv , do you see a user case, that there is need to support PVH-style direct kernel boot? 
It is not supported today. And I think it could be added as an option, if there is a solid use case.

dimakuv, 2022-02-16:
> In case of TDX, the firmware (TDShim) is the one implementing the protocol. That means even if Cloud Hypervisor doesn't support Linux Boot Protocol, it is simply copying the kernel somewhere in memory so that the TDShim can handle the filling of zero page and the kernel boot parameters.

Thanks @sboeuf, this is the part that I was missing (that non-TDX and TDX flows differ in the preparation-for-kernel-entry). In retrospect, this is obvious, since TD-Shim cannot trust the VMM to correctly handle the protocol.

> @dimakuv , do you see a user case, that there is need to support PVH-style direct kernel boot?
> It is not supported today. And I think it could be added as an option, if there is a solid use case.

@jyao1 No, currently I have no solid use case. I was just curious about the seeming incompatibility of TD-Shim and CloudHypervisor.

closed 2022-02-16 by dimakuv