confidential-containers/td-shim #99: Does TD-Shim require patches to QEMU/KVM?

dimakuv, 2022-02-16:
Of course, to test TD workloads, both QEMU and KVM must be TDX-enlightened.

But is it enough to use QEMU from https://github.com/intel/qemu-tdx/ and KVM from https://github.com/intel/tdx/tree/kvm repos?

Or are there any additional patches that must be applied specifically for TD-Shim?

Also, are there known "good stable" releases/commit hashes of QEMU and KVM that are recommended for use with TD-Shim?

When reading the TD-Shim spec, I got a feeling that QEMU/KVM do not need to be aware of TD-Shim. In other words, TD-Shim is more or less a drop-in replacement for TDVF. Is my understanding correct?

gaojiaqi7, 2022-02-17:
hi @dimakuv 

> Of course, to test TD workloads, both QEMU and KVM must be TDX-enlightened.
> 
> But is it enough to use QEMU from https://github.com/intel/qemu-tdx/ and KVM from https://github.com/intel/tdx/tree/kvm repos?
> 
> Or are there any additional patches that must be applied specifically for TD-Shim?
> 

There's no need to apply additional patches on KVM, but it is required to make some changes on QEMU (https://github.com/intel/qemu-tdx/) to support full features of td-shim such as ACPI tables and kernel direct boot. Since currently TDVF is using fw-cfg mechanism to get ACPI tables/ kernel/ command line from QEMU, and TD-Shim defines another way which is to pass these things through HOB (https://github.com/confidential-containers/td-shim/blob/staging/doc/tdshim_spec.md#part-ii---vmm--td-shim-interface)

closed 2022-02-17 by jyao1

dimakuv, 2022-02-17:
> There's no need to apply additional patches on KVM, but it is required to make some changes on QEMU (https://github.com/intel/qemu-tdx/) to support full features of td-shim such as ACPI tables and kernel direct boot.

These TD-Shim-specific patches to QEMU are available somewhere, so I could fetch them?

gaojiaqi7, 2022-02-17:
> These TD-Shim-specific patches to QEMU are available somewhere, so I could fetch them?

There‘re no such patches for QEMU as I know🙂

dimakuv, 2022-02-17:
So you're using unmodified QEMU from https://github.com/intel/qemu-tdx/ to test a subset of functionality in TD-Shim?

But you envision that to test the whole functionality of TD-Shim, QEMU will be patched to support this functionality (like VMM providing HOBs to TD-Shim)?