confidential-containers/td-shim #103: Failed to enable libfuzzer

haowqs, 2022-02-17:
Rust has the cargo-fuzz library to call libfuzzer
I want to test td-loader.

```
$ cd td-loader
$ cargo fuzz init
$ cargo fuzz add elf
// td-loader/fuzz/fuzz_targets/elf.rs
#![no_main]
use libfuzzer_sys::fuzz_target;

use td_loader::elf;

fuzz_target!(|data: &[u8]| {
    // fuzzed code goes here
    let _ = elf::is_elf(data);
});
```
run elf
```
$ cargo fuzz run elf
fuzz@fuzz:~/td-shim/td-loader$ cargo fuzz run elf
   Compiling proc-macro2 v1.0.36
   Compiling unicode-xid v0.2.2
   Compiling syn v1.0.86
   Compiling log v0.4.14
   Compiling cc v1.0.73
   Compiling cfg-if v1.0.0
   Compiling arbitrary v1.1.0
   Compiling once_cell v1.9.0
   Compiling quote v1.0.15
   Compiling libfuzzer-sys v0.4.2
   Compiling scroll_derive v0.10.5
   Compiling scroll v0.10.2
   Compiling td-loader v0.1.0 (/home/fuzz/td-shim/td-loader)
warning: unknown pass `sancov-module`, ignoring

warning: `td-loader` (lib) generated 1 warning
   Compiling td-loader-fuzz v0.0.0 (/home/fuzz/td-shim/td-loader/fuzz)
warning: `td-loader-fuzz` (bin "elf") generated 1 warning (1 duplicate)
    Finished release [optimized] target(s) in 25.09s
warning: unknown pass `sancov-module`, ignoring

warning: `td-loader` (lib) generated 1 warning
warning: `td-loader-fuzz` (bin "elf") generated 1 warning (1 duplicate)
    Finished release [optimized] target(s) in 0.01s
     Running `fuzz/target/x86_64-unknown-linux-gnu/release/elf -artifact_prefix=/home/fuzz/td-shim/td-loader/fuzz/artifacts/elf/ /home/fuzz/td-shim/td-loader/fuzz/corpus/elf`
INFO: Running with entropic power schedule (0xFF, 100).
INFO: Seed: 383398225
INFO:        9 files found in /home/fuzz/td-shim/td-loader/fuzz/corpus/elf
INFO: -max_len is not provided; libFuzzer will not generate inputs larger than 4096 bytes
INFO: seed corpus: files: 9 min: 1b max: 40b total: 145b rss: 29Mb
#10     INITED exec/s: 0 rss: 30Mb
ERROR: no interesting inputs were found. Is the code instrumented for coverage? Exiting.
────────────────────────────────────────────────────────────────────────────────

Error: Fuzz target exited with exit status: 1
```

haowqs, 2022-02-17:
I modified the rust-toolchain file and switched the rust version to nightly.
runed result
```
fuzz@td-fuzz:~/td-shim/td-loader$ cargo fuzz run elf
   Compiling proc-macro2 v1.0.36
   Compiling unicode-xid v0.2.2
   Compiling syn v1.0.86
   Compiling cc v1.0.73
   Compiling log v0.4.14
   Compiling cfg-if v1.0.0
   Compiling once_cell v1.9.0
   Compiling arbitrary v1.1.0
   Compiling quote v1.0.15
   Compiling libfuzzer-sys v0.4.2
   Compiling scroll_derive v0.10.5
   Compiling scroll v0.10.2
   Compiling td-loader v0.1.0 (/home/fuzz/td-shim/td-loader)
   Compiling td-loader-fuzz v0.0.0 (/home/fuzz/td-shim/td-loader/fuzz)
    Finished release [optimized] target(s) in 29.68s
    Finished release [optimized] target(s) in 0.01s
     Running `fuzz/target/x86_64-unknown-linux-gnu/release/elf -artifact_prefix=/home/fuzz/td-shim/td-loader/fuzz/artifacts/elf/ /home/fuzz/td-shim/td-loader/fuzz/corpus/elf`
INFO: Running with entropic power schedule (0xFF, 100).
INFO: Seed: 2269920554
INFO: Loaded 1 modules   (10618 inline 8-bit counters): 10618 [0x55bf684c7140, 0x55bf684c9aba), 
INFO: Loaded 1 PC tables (10618 PCs): 10618 [0x55bf684c9ac0,0x55bf684f3260), 
INFO:        9 files found in /home/fuzz/td-shim/td-loader/fuzz/corpus/elf
INFO: -max_len is not provided; libFuzzer will not generate inputs larger than 4096 bytes
INFO: seed corpus: files: 9 min: 1b max: 40b total: 145b rss: 30Mb
#10     INITED cov: 19 ft: 20 corp: 3/13b exec/s: 0 rss: 32Mb
#2097152        pulse  cov: 19 ft: 20 corp: 3/13b lim: 4096 exec/s: 699050 rss: 189Mb
#4194304        pulse  cov: 19 ft: 20 corp: 3/13b lim: 4096 exec/s: 599186 rss: 346Mb
```

haowqs, 2022-02-21:
rust-version|libfuzzer|td-shim|result
:-:|:-:|:-:|:-:
nightly-2020-02-08|0.11.0|staging|no (td-shim fails to compile in this version)
nightly-2021-08-20|0.10.2|staging|yes
nightly-2020-02-08|0.11.0|staging|no (libfuzzer failed to start)

jyao1, 2022-02-21:
Then please enable libfuzzer 0.10.2 at first.
Once we upgrade td-shim to be compatible with nightly-2020-02-08, we can update libfuzzer to 0.11.0.

haowqs, 2022-04-28:
The rust version has been upgraded, so 0.11.0 can be uesd.

closed 2022-04-28 by haowqs