confidential-containers/td-shim #104: staging: The branch is not buidable due to broken external dependencies (on February 19th, 2022)

fidencio, 2022-02-19:
```
[ffidenci@ffidenci-desk td-shim]$ export CC=clang; export AR=llvm-ar
[ffidenci@ffidenci-desk td-shim]$ 
[ffidenci@ffidenci-desk td-shim]$ cargo xbuild -p td-shim --target x86_64-unknown-uefi --release --features=main,tdx,boot-kernel
WARNING: There is no root package to read the cargo-xbuild config from.
   Compiling core v0.0.0 (/home/ffidenci/.rustup/toolchains/nightly-2021-08-20-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
   Compiling compiler_builtins v0.1.49
   Compiling rustc-std-workspace-core v1.99.0 (/home/ffidenci/.rustup/toolchains/nightly-2021-08-20-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling alloc v0.0.0 (/tmp/cargo-xbuildwzHzo5)
    Finished release [optimized] target(s) in 10.77s
   Compiling proc-macro2 v1.0.36
   Compiling unicode-xid v0.2.2
   Compiling syn v1.0.86
   Compiling log v0.4.14
   Compiling cfg-if v1.0.0
   Compiling scopeguard v1.1.0
   Compiling cc v1.0.70 (https://github.com/jyao1/cc-rs.git?branch=uefi_support#3b0362f9)
   Compiling bitflags v1.3.2
   Compiling spin v0.5.2
   Compiling x86_64 v0.14.6
   Compiling bit_field v0.10.1
   Compiling volatile v0.4.4
   Compiling x86 v0.44.0
   Compiling r-efi v3.2.0
   Compiling bitfield v0.13.2
   Compiling untrusted v0.7.1
   Compiling byteorder v1.4.3
   Compiling raw-cpuid v10.2.0
   Compiling lock_api v0.4.6
   Compiling lazy_static v1.4.0
   Compiling spin v0.9.2
   Compiling spinning_top v0.2.4
   Compiling linked_list_allocator v0.9.1
   Compiling ring v0.16.20 (https://github.com/jyao1/ring.git?branch=uefi_support#3e042088)
   Compiling quote v1.0.15
error: failed to run custom build command for `ring v0.16.20 (https://github.com/jyao1/ring.git?branch=uefi_support#3e042088)`

Caused by:
  process didn't exit successfully: `/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/release/build/ring-71a085a804d92895/build-script-build` (exit status: 101)
  --- stderr
  running "perl" "crypto/fipsmodule/aes/asm/aesni-x86_64.pl" "nasm" "/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/aesni-x86_64-nasm.asm"
  running "perl" "crypto/fipsmodule/aes/asm/vpaes-x86_64.pl" "nasm" "/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/vpaes-x86_64-nasm.asm"
  running "perl" "crypto/fipsmodule/bn/asm/x86_64-mont.pl" "nasm" "/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/x86_64-mont-nasm.asm"
  running "perl" "crypto/fipsmodule/bn/asm/x86_64-mont5.pl" "nasm" "/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/x86_64-mont5-nasm.asm"
  running "perl" "crypto/chacha/asm/chacha-x86_64.pl" "nasm" "/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/chacha-x86_64-nasm.asm"
  running "perl" "crypto/fipsmodule/ec/asm/p256-x86_64-asm.pl" "nasm" "/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/p256-x86_64-asm-nasm.asm"
  running "perl" "crypto/fipsmodule/modes/asm/aesni-gcm-x86_64.pl" "nasm" "/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/aesni-gcm-x86_64-nasm.asm"
  running "perl" "crypto/fipsmodule/modes/asm/ghash-x86_64.pl" "nasm" "/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/ghash-x86_64-nasm.asm"
  running "perl" "crypto/fipsmodule/sha/asm/sha512-x86_64.pl" "nasm" "/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/sha512-x86_64-nasm.asm"
  running "perl" "crypto/cipher_extra/asm/chacha20_poly1305_x86_64.pl" "nasm" "/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/chacha20_poly1305_x86_64-nasm.asm"
  running "perl" "crypto/fipsmodule/sha/asm/sha512-x86_64.pl" "nasm" "/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/sha256-x86_64-nasm.asm"
  running "nasm" "-o" "/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/aesni-x86_64-nasm.o" "-f" "win64" "-Xgnu" "-gcv8" "/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/aesni-x86_64-nasm.asm"
  thread 'main' panicked at 'failed to execute ["nasm" "-o" "/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/aesni-x86_64-nasm.o" "-f" "win64" "-Xgnu" "-gcv8" "/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/aesni-x86_64-nasm.asm"]: No such file or directory (os error 2)', /home/ffidenci/.cargo/git/checkouts/ring-1836a5bf48f90a2a/3e04208/build.rs:660:9
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: build failed
```

fidencio, 2022-02-19:
I don't think there's much to do here apart from wait till it the external dependency gets fixed, but at least the issue report raises awareness and documents when breakages happen, regardless of those being on external dependencies or not.

jiangliu, 2022-02-19:
I have thought about to maintain local patches for cc/ring, but that will introduce "git submodule",  another annoying dependency:(

jiangliu, 2022-02-19:
Have you install the "nasm" package? I remember the same command succeeds on my side.

jyao1, 2022-02-19:
I can build successfully with below:
```
cargo xbuild -p td-shim --target x86_64-unknown-uefi --release --features=main,tdx

cargo xbuild -p td-payload --target x86_64-unknown-uefi --release --features=main,tdx

cargo xbuild -p td-payload --target devtools/rustc-targets/x86_64-unknown-none.json --release --features=main,tdx
```

With a warning is shown 
```
warning: Linking two modules of different data layouts: '' is 'e-m:e-p270:32:32-p271:32:32-p272:64:64-i64:64-f80:128-n8:16:32:64-S128' whereas 'td_payload.0b8d0db6-cgu.11' is 'e-m:e-i64:64-f80:128-n8:16:32:64-S128'
```

Would you please double check if you are following readme?
```
2. Install [NASM](https://www.nasm.us/)
Please make sure nasm can be found in PATH.

3. Install LLVM
Please make sure clang can be found in PATH.

Set env:
set CC=clang
set AR=llvm-ar
```

jiangliu, 2022-02-19:
Confirmed, with a docker image built from devtools/dev_container/Dockerfile, the command succeeds:)
```
cargo xbuild -p td-shim --target x86_64-unknown-uefi --release --features=main,tdx,boot-kernel

```

fidencio, 2022-02-21:
@jiangliu @jyao1, please, I'd really prefer if this would not be closed as "work for me" without we understand what's actually happening here, as more people can hit this one, even if by a misconfiguration on their side.

```
[ffidenci@ffidenci-desk td-shim]$ echo $CC
clang

[ffidenci@ffidenci-desk td-shim]$ echo $AR
llvm-ar

[ffidenci@ffidenci-desk td-shim]$ whereis nasm
nasm: /usr/bin/nasm /usr/share/man/man1/nasm.1.gz

[ffidenci@ffidenci-desk td-shim]$ echo $PATH
/usr/local/go/bin:/usr/local/go/bin:/home/ffidenci/.local/bin:/home/ffidenci/bin:/home/ffidenci/.cargo/bin:/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/home/ffidenci/go/bin:/home/ffidenci/go/bin

[ffidenci@ffidenci-desk td-shim]$ cargo xbuild -p td-shim --target x86_64-unknown-uefi --release --features=main,tdx
WARNING: There is no root package to read the cargo-xbuild config from.
   Compiling ring v0.16.20 (https://github.com/jyao1/ring.git?branch=uefi_support#3e042088)
   Compiling scroll_derive v0.10.5
   Compiling synstructure v0.12.6
error: failed to run custom build command for `ring v0.16.20 (https://github.com/jyao1/ring.git?branch=uefi_support#3e042088)`

Caused by:
  process didn't exit successfully: `/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/release/build/ring-71a085a804d92895/build-script-build` (exit status: 101)
  --- stdout
  OPT_LEVEL = Some("3")
  TARGET = Some("x86_64-unknown-uefi")
  HOST = Some("x86_64-unknown-linux-gnu")
  CC_x86_64-unknown-uefi = None
  CC_x86_64_unknown_uefi = None
  TARGET_CC = None
  CC = Some("clang")
  CFLAGS_x86_64-unknown-uefi = None
  CFLAGS_x86_64_unknown_uefi = None
  TARGET_CFLAGS = None
  CFLAGS = None
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some("false")

  --- stderr
  running "clang" "-O3" "-ffunction-sections" "-fdata-sections" "--target=x86_64-unknown-windows-gnu" "-I" "include" "-Wall" "-Wextra" "-std=c1x" "-Wbad-function-cast" "-Wnested-externs" "-Wstrict-prototypes" "-pedantic" "-pedantic-errors" "-Wall" "-Wextra" "-Wcast-align" "-Wcast-qual" "-Wconversion" "-Wenum-compare" "-Wfloat-equal" "-Wformat=2" "-Winline" "-Winvalid-pch" "-Wmissing-field-initializers" "-Wmissing-include-dirs" "-Wredundant-decls" "-Wshadow" "-Wsign-compare" "-Wsign-conversion" "-Wundef" "-Wuninitialized" "-Wwrite-strings" "-fno-strict-aliasing" "-fvisibility=hidden" "-g3" "-nostdlibinc" "-ffreestanding" "-Werror" "-DNDEBUG" "-DGFp_NOSTDLIBINC=1" "-c" "-o/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/aes_nohw.o" "crypto/fipsmodule/aes/aes_nohw.c"
  thread 'main' panicked at 'failed to execute ["clang" "-O3" "-ffunction-sections" "-fdata-sections" "--target=x86_64-unknown-windows-gnu" "-I" "include" "-Wall" "-Wextra" "-std=c1x" "-Wbad-function-cast" "-Wnested-externs" "-Wstrict-prototypes" "-pedantic" "-pedantic-errors" "-Wall" "-Wextra" "-Wcast-align" "-Wcast-qual" "-Wconversion" "-Wenum-compare" "-Wfloat-equal" "-Wformat=2" "-Winline" "-Winvalid-pch" "-Wmissing-field-initializers" "-Wmissing-include-dirs" "-Wredundant-decls" "-Wshadow" "-Wsign-compare" "-Wsign-conversion" "-Wundef" "-Wuninitialized" "-Wwrite-strings" "-fno-strict-aliasing" "-fvisibility=hidden" "-g3" "-nostdlibinc" "-ffreestanding" "-Werror" "-DNDEBUG" "-DGFp_NOSTDLIBINC=1" "-c" "-o/home/ffidenci/go/src/github.com/confidential-containers/td-shim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/aes_nohw.o" "crypto/fipsmodule/aes/aes_nohw.c"]: No such file or directory (os error 2)', /home/ffidenci/.cargo/git/checkouts/ring-1836a5bf48f90a2a/3e04208/build.rs:660:9
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: build failed
```

As you can see, CC is set to `clang`, AR is set to `llvm-ar`, `nasm` is installed and it is in the `$PATH`.  What else may I be missing here?

jiangliu, 2022-02-21:
> 

Could you please help to confirm whether clang is installed on your system?
If I set CC=clang1, (which is apparently doesn't exist on my system), I got the same error as yours. So seems clang is missing on your system.
```
root@2e93f4f65f60:/tdshim# clang
clang: error: no input files
root@2e93f4f65f60:/tdshim# clang1
bash: clang1: command not found

root@2e93f4f65f60:/tdshim# echo $CC
clang1
root@2e93f4f65f60:/tdshim# cargo xbuild -p td-shim --target x86_64-unknown-uefi --release --features=main,tdx
WARNING: There is no root package to read the cargo-xbuild config from.
   Compiling ring v0.16.20 (https://github.com/jyao1/ring.git?branch=uefi_support#3e042088)
error: failed to run custom build command for `ring v0.16.20 (https://github.com/jyao1/ring.git?branch=uefi_support#3e042088)`

Caused by:
  process didn't exit successfully: `/tdshim/target/release/build/ring-71a085a804d92895/build-script-build` (exit status: 101)
  --- stdout
  OPT_LEVEL = Some("3")
  TARGET = Some("x86_64-unknown-uefi")
  HOST = Some("x86_64-unknown-linux-gnu")
  CC_x86_64-unknown-uefi = None
  CC_x86_64_unknown_uefi = None
  TARGET_CC = None
  CC = Some("clang1")
  CFLAGS_x86_64-unknown-uefi = None
  CFLAGS_x86_64_unknown_uefi = None
  TARGET_CFLAGS = None
  CFLAGS = None
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some("false")

  --- stderr
  running "clang1" "-O3" "-ffunction-sections" "-fdata-sections" "--target=x86_64-unknown-windows-gnu" "-I" "include" "-Wall" "-Wextra" "-std=c1x" "-Wbad-function-cast" "-Wnested-externs" "-Wstrict-prototypes" "-pedantic" "-pedantic-errors" "-Wall" "-Wextra" "-Wcast-align" "-Wcast-qual" "-Wconversion" "-Wenum-compare" "-Wfloat-equal" "-Wformat=2" "-Winline" "-Winvalid-pch" "-Wmissing-field-initializers" "-Wmissing-include-dirs" "-Wredundant-decls" "-Wshadow" "-Wsign-compare" "-Wsign-conversion" "-Wundef" "-Wuninitialized" "-Wwrite-strings" "-fno-strict-aliasing" "-fvisibility=hidden" "-g3" "-nostdlibinc" "-ffreestanding" "-Werror" "-DNDEBUG" "-DGFp_NOSTDLIBINC=1" "-c" "-o/tdshim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/aes_nohw.o" "crypto/fipsmodule/aes/aes_nohw.c"
  thread 'main' panicked at 'failed to execute ["clang1" "-O3" "-ffunction-sections" "-fdata-sections" "--target=x86_64-unknown-windows-gnu" "-I" "include" "-Wall" "-Wextra" "-std=c1x" "-Wbad-function-cast" "-Wnested-externs" "-Wstrict-prototypes" "-pedantic" "-pedantic-errors" "-Wall" "-Wextra" "-Wcast-align" "-Wcast-qual" "-Wconversion" "-Wenum-compare" "-Wfloat-equal" "-Wformat=2" "-Winline" "-Winvalid-pch" "-Wmissing-field-initializers" "-Wmissing-include-dirs" "-Wredundant-decls" "-Wshadow" "-Wsign-compare" "-Wsign-conversion" "-Wundef" "-Wuninitialized" "-Wwrite-strings" "-fno-strict-aliasing" "-fvisibility=hidden" "-g3" "-nostdlibinc" "-ffreestanding" "-Werror" "-DNDEBUG" "-DGFp_NOSTDLIBINC=1" "-c" "-o/tdshim/target/x86_64-unknown-uefi/release/build/ring-6d4280e73a5949bb/out/aes_nohw.o" "crypto/fipsmodule/aes/aes_nohw.c"]: No such file or directory (os error 2)', /root/.cargo/git/checkouts/ring-1836a5bf48f90a2a/3e04208/build.rs:660:9
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

fidencio, 2022-02-21:
@jiangliu, I really am ashamed. /o\

Looking my history here, at first I built everything inside my toolbox (a Fedora 35 container) and for some reason I totally erased that from my mind.  That's on me, my fault. /o\.

Of course, I had all the dependencies installed inside the toolbox, but not on my host OS ... and here we're right now.
Sorry for the noise, and I truly am embarassed about that. /o\.

closed 2022-02-21 by fidencio

jodh-intel, 2022-02-21:
@fidencio - honestly, it's useful to have issues like this as it's an easy mistake to make and having this issue makes the solution discoverable. It's bound to help someone in the future with the same problem!

Honestly, we should make `build.rs` check for the required tooling and fail early rather than the current behaviour imho.

jiangliu, 2022-02-22:
> @fidencio - honestly, it's useful to have issues like this as it's an easy mistake to make and having this issue makes the solution discoverable. It's bound to help someone in the future with the [...]

That's a great idea, but I do suggest to use dev container and Makefile, they are reproducible:)

jodh-intel, 2022-02-22:
@jiangliu's - I've only just seen this comment but I've just raised #116 to avoid issues like this.