confidential-containers/td-shim #109: Clarify why shim won't build with GCC

jodh-intel, 2022-02-21:
https://github.com/confidential-containers/td-shim/tree/staging shows that the shim must be built with `clang`. However, we should:

- Explain why the default compiler oin most Linux distros (GCC) won't work.
- Understand and ideally fix the GCC issue.

xiaoyuxlu, 2022-03-11:
> Explain why the default compiler oin most Linux distros (GCC) won't work.

My understanding is that
Rustc target of `TD-shim` is `x86_64-unknown-uefi` (PE format). Linux GCC generates ELF objs, but `rustc target x86_64-unknown-uefi` generates PE objs. So link failed. Maybe x86_64-w64-mingw32-gcc would work. 

> Understand and ideally fix the GCC issue.

I think it's very difficult at the moment. Unless:
the distribution (GCC) supports compiling the PE format. 
Or
We choose another rust-toolchain, like: x86_64-unknown-none(tier3)(elf format).  But in this way we can't solve the problem that the `cl.exe` compiler can't build the elf obj on windows. 😭 

So we choose `clang`.

ariel-adam, 2022-10-11:
@jodh-intel is this issue still relevant or can be closed? 
If it's still relevant to what release do you think we should map it to (mid-November, end-December, mid-February etc...)?

jodh-intel, 2022-10-11:
Closing since @xiaoyuxlu has explained this limitation.

closed 2022-10-11 by jodh-intel