confidential-containers/td-shim #110: build script should check dependencies and fail early

jodh-intel, 2022-02-21:
Building the shim is currently somewhat awkward since you need:

- clang ([GCC won't work](#109)).
- nasm
- `cargo xbuild`

Admittedly, this _is_ documented in https://github.com/confidential-containers/td-shim/blob/staging/README.md, but we should make it easier for users by checking for the expected commands in `build.rs` so that the build can fail early (rather than letting it chug on and die in flames at some later stage).

Related:

- https://github.com/confidential-containers/td-shim/pull/2#issuecomment-977991422
- https://github.com/confidential-containers/td-shim/pull/2#discussion_r759137136

jyao1, 2022-02-22:
@jodh-intel, can you take this and propose a patch?

jodh-intel, 2022-02-22:
Yes, I can look at this, but I don't have privs to assign this issue to myself.

@sameo - could you possibly assign to me or grant me those privs?

PR #116 (2022-02-28): td-shim: Check build tooling
Although the required tooling is documented in the README, it's too easy
to end up with a failed build due to missing tooling.

Add a new function to the build script to check for all dependencies and
fail with helpful messages on error.

Fixes: #110.
PR #133 (2022-03-10): td-shim: Make build script check for required commands
Although the required tooling is documented in the README, it's too easy
to end up with a failed build due to missing tooling.

Add a new function to the build script to check for all dependencies and
fail with helpful messages on error.

Fixes: #110.
commit 9a4284a8 (2022-03-10): td-shim: Make build script check for required commands

Although the required tooling is documented in the README, it's too easy
to end up with a failed build due to missing tooling.

Add a new function to the build script to check for all dependencies and
fail with helpful messages on error.

Fixes: #110.
closed 2022-03-10 by jiangliu