confidential-containers/td-shim #114: Add more tools to devtools/dev_container/Dockerfile and Makefile

jyao1, 2022-02-22:
Ref: https://github.com/confidential-containers/td-shim/pull/107#issuecomment-1047388165

We should add more tools to Dockerfile and Makefile, such as fuzzing (afl, libfuzzer, rudra, etc)

haowqs, 2022-02-23:
I want to add these tools.

haowqs, 2022-02-24:
@jiangliu 
hi，I add tools to dockerfile and makefile, do I just need to implement these tools, or should I add both dockerfile and makefile?

haowqs, 2022-02-24:
|           |   dockerfile or makefile   |         dockerfile and makefile         |    dockerfile and makefile    |    dockerfile and makefile    |
| :-------: | :------------------------: | :-------------------------------------: | :---------------------------: | :---------------------------: |
|    afl/libfuzzer/rudra    |            yes             |                   yes                   | yes(dockerfile call makefile) | yes(makefile call dockerfile) |

jiangliu, 2022-02-24:
> @jiangliu hi，I add tools to dockerfile and makefile, do I just need to implement these tools, or should I add both dockerfile and makefile?

Dockerfile is to install required tools/components, makefile is to actually execute tests.

PR #124 (2022-03-04): Add tools to dockerfile and makefile
Fix: #114 

I add environment in devtools/dev_container/dockerfile about fuzzing tools afl and libfuzzer and static analysis tool rudra.

I add docker call in devtools/makefile, start dockerfile run afl and libfuzzer and rudra.

Modify the changes in sh_script/rudra.sh.