confidential-containers/td-shim #135: Enhance td-shim to support multi-arch

jiangliu, 2022-03-08:
Currently the td-shim implementation has no support for AMD SEV/CSV etc, it would be great to enhance the td-shim code to prepare for mutli-arch support.

jyao1, 2022-03-08:
I think we need AMD SEV/CSV expert to take a look at this.

jiangliu, 2022-03-10:
@fitzthum any comments?

fitzthum, 2022-03-10:
I think it would be great to have a firmware that works on all platforms. We've been using OVMF with SEV mainly because that's where AMD have contributed their SEV support patches. Those patches can give us some ideas about how to support SEV elsewhere. Since SEV has 3 different features (SEV, -ES, SNP) we might be able to ease our way into supporting it.

The main question for me is what would `td-shim` (which we might want to rename if it ends up supporting more stuff) give us over OVMF? OVMF is already relatively small (especially if you use the AmdSev package), so I'm not sure that size is significant. What do you think are the true benefits in terms of footprint and boot time? 

Although OVMF is the de-facto solution and has by far the most comprehensive SEV support, there are some things that I don't love about it. Mainly, that it is a relatively large and confusing codebase with a weird build system that isn't reproducible. It seems like we could sidestep all of those issues here.

I can't promise any patches, but I will at least try to look through the original SEV OVMF patches and see what we would need to do to support basic SEV. 

CC: @larrydewey @dubek

ariel-adam, 2022-10-11:
@jiangliu is this issue still relevant or can be closed? 
If it's still relevant to what release do you think we should map it to (mid-November, end-December, mid-February etc...)?

jinankjain, 2023-09-19:
@fitzthum @jiangliu Is anyone still actively working on it?

fitzthum, 2023-09-20:
I am not aware of anyone working on supporting other platforms with td-shim. Might be a good project for someone.