confidential-containers/td-shim #136: Enhance td-shim/ResetVector to support test on non-TDX hardware platfroms

jiangliu, 2022-03-08:
It help ease the development if we could test td-shim on non-TDX hardware platforms. Of course, all TDX related functionality will be disabled on non-TDX hardware platform.
Basically we need to enhance ResetVector to support boot from real mode and then switch protected as what expected by TDX.

jyao1, 2022-03-08:
Agree from high level.

Question: What do you think the **interface** between hypervisor in non-tdx shim ?
Option 1: We reuse the normal non-tdx interface. E.g. Shim code runs from real mode. Shim gets memory info from CMOS.
Option 2: We reuse the td-shim interface. E.g. Shim code runs from protected mode. Shim gets the memory info via hob in RCX.

jiangliu, 2022-03-10:
I like option2, we should share as much code as possible with tdx.
Actually we need a shim for td-shim to setup the environment defined/expected by tdx

ariel-adam, 2022-10-11:
@jiangliu is this issue still relevant or can be closed? 
If it's still relevant to what release do you think we should map it to (mid-November, end-December, mid-February etc...)?