confidential-containers/td-shim #159: td-shim: Booting up a VM with 14+ vCPUs fail due to: "VmBoot(PopulateHob(GuestMemoryWriteHob(InvalidGuestAddress(GuestAddress(4278464512)))))"

fidencio, 2022-03-17:
When booting up a TDX VM using Cloud Hypervisor with td-shim, using kata-containers guest and kernel images, it fails.

The components used are:
* cloud-hypervisor: v22.1, built with `--tdx` support
  * http://jenkins.katacontainers.io/job/kata-containers-2.0-clh-tdx-x86_64/
* guest kernel: vmlinuz-tdx-guest-v5.15-4-90-tdx
  * http://jenkins.katacontainers.io/job/kata-containers-2.0-kernel-tdx-x86_64-nightly/lastSuccessfulBuild/artifact/artifacts/vmlinuz-tdx-guest-v5.15-4-90-tdx
* guest image: The one provided by the kata-containers official 2.4.0-rc0 release
  * https://github.com/kata-containers/kata-containers/releases/download/2.4.0-rc0/kata-static-2.4.0-rc0-x86_64.tar.xz
    * It's located under ./opt/kata/share/kata-containers/kata-containers.img
* td-shim: 85b71f2316b2680b815823420909f91ee53bde25, from the staging branch.
  * Built with the following commands:
    ```
    cargo xbuild -p td-shim --target x86_64-unknown-uefi --release --features=main,tdx,boot-kernel
    cargo xbuild -p td-payload --target x86_64-unknown-uefi --release --features=main,tdx
    cargo run -p td-shim-tools --features=td-shim/default,td-shim/tdx,boot-kernel --bin td-shim-ld -- target/x86_64-unknown-uefi/release/ResetVector.bin target/x86_64-unknown-uefi/release/td-shim.efi target/x86_64-unknown-uefi/release/td-payload.efi -o target/x86_64-unknown-uefi/release/final-pe.bin
    ```

And the command line used to spawn the guest:
```
cloud-hypervisor --tdx firmware=final-pe.bin --kernel vmlinuz-tdx-guest-v5.15-4-90-tdx --cmdline "root=/dev/vda1 sonole=hvc0 rw force_tdx_guest tdx_disable_filter" --cpus boot=14 --memory size=2G --disk path=kata-containers.img -
```

<details>
<summary>Here you can find the full log:</summary>

```
$ ./cloud-hypervisor --tdx firmware=final-pe.bin --kernel vmlinuz-tdx-guest-v5.15-4-90-tdx --cmdline "root=/dev/vda1 sonole=hvc0 rw force_tdx_guest tdx_disable_filter" --cpus boot=14 --memory size=4G --disk path=kata-containers.img -v
cloud-hypervisor: 4.077994ms: <vmm> INFO:vmm/src/lib.rs:1578 -- API request event: VmCreate(Mutex { data: VmConfig { cpus: CpusConfig { boot_vcpus: 14, max_vcpus: 14, topology: None, kvm_hyperv: false, max_phys_bits: 46, affinity: None }, memory: MemoryConfig { size: 4294967296, mergeable: false, hotplug_method: Acpi, hotplug_size: None, hotplugged_size: None, shared: false, hugepages: false, hugepage_size: None, prefault: false, zones: None }, kernel: Some(KernelConfig { path: "vmlinuz-tdx-guest-v5.15-4-90-tdx" }), initramfs: None, cmdline: CmdlineConfig { args: "root=/dev/vda1 sonole=hvc0 rw force_tdx_guest tdx_disable_filter" }, disks: Some([DiskConfig { path: Some("kata-containers.img"), readonly: false, direct: false, iommu: false, num_queues: 1, queue_size: 128, vhost_user: false, vhost_socket: None, poll_queue: true, rate_limiter_config: None, id: None, disable_io_uring: false, pci_segment: 0 }]), net: None, rng: RngConfig { src: "/dev/urandom", iommu: false }, balloon: None, fs: None, pmem: None, serial: ConsoleConfig { file: None, mode: Null, iommu: false }, console: ConsoleConfig { file: None, mode: Tty, iommu: false }, devices: None, user_devices: None, vsock: None, iommu: false, sgx_epc: None, numa: None, watchdog: false, tdx: Some(TdxConfig { firmware: "final-pe.bin" }), platform: None }, poisoned: false, .. }, Sender { .. })
cloud-hypervisor: 5.205661ms: <vmm> INFO:vmm/src/lib.rs:1578 -- API request event: VmBoot(Sender { .. })
cloud-hypervisor: 8.254521ms: <vmm> INFO:vmm/src/memory_manager.rs:1475 -- Creating userspace mapping: 0 -> 7fc4c554a000 c0000000, slot 0
cloud-hypervisor: 12.545088ms: <vmm> INFO:vmm/src/memory_manager.rs:1509 -- Created userspace mapping: 0 -> 7fc4c554a000 c0000000
cloud-hypervisor: 12.594762ms: <vmm> INFO:vmm/src/memory_manager.rs:1475 -- Creating userspace mapping: 100000000 -> 7fc48554a000 40000000, slot 1
cloud-hypervisor: 13.312823ms: <vmm> INFO:vmm/src/memory_manager.rs:1509 -- Created userspace mapping: 100000000 -> 7fc48554a000 40000000
cloud-hypervisor: 13.364081ms: <vmm> INFO:vmm/src/vm.rs:567 -- Booting VM from config: Mutex { data: VmConfig { cpus: CpusConfig { boot_vcpus: 14, max_vcpus: 14, topology: None, kvm_hyperv: false, max_phys_bits: 46, affinity: None }, memory: MemoryConfig { size: 4294967296, mergeable: false, hotplug_method: Acpi, hotplug_size: None, hotplugged_size: None, shared: false, hugepages: false, hugepage_size: None, prefault: false, zones: None }, kernel: Some(KernelConfig { path: "vmlinuz-tdx-guest-v5.15-4-90-tdx" }), initramfs: None, cmdline: CmdlineConfig { args: "root=/dev/vda1 sonole=hvc0 rw force_tdx_guest tdx_disable_filter" }, disks: Some([DiskConfig { path: Some("kata-containers.img"), readonly: false, direct: false, iommu: false, num_queues: 1, queue_size: 128, vhost_user: false, vhost_socket: None, poll_queue: true, rate_limiter_config: None, id: None, disable_io_uring: false, pci_segment: 0 }]), net: None, rng: RngConfig { src: "/dev/urandom", iommu: false }, balloon: None, fs: None, pmem: None, serial: ConsoleConfig { file: None, mode: Null, iommu: false }, console: ConsoleConfig { file: None, mode: Tty, iommu: false }, devices: None, user_devices: None, vsock: None, iommu: false, sgx_epc: None, numa: None, watchdog: false, tdx: Some(TdxConfig { firmware: "final-pe.bin" }), platform: None }, poisoned: false, .. }
cloud-hypervisor: 14.326287ms: <vmm> INFO:vmm/src/pci_segment.rs:92 -- Adding PCI segment: id=0, PCI MMIO config address: 0xe8000000, device area [0x140000000-0x3fff3fffffff
cloud-hypervisor: 14.804367ms: <vmm> INFO:vmm/src/device_manager.rs:1959 -- Creating virtio-block device: DiskConfig { path: Some("kata-containers.img"), readonly: false, direct: false, iommu: false, num_queues: 1, queue_size: 128, vhost_user: false, vhost_socket: None, poll_queue: true, rate_limiter_config: None, id: Some("_disk0"), disable_io_uring: false, pci_segment: 0 }
cloud-hypervisor: 15.300825ms: <vmm> INFO:vmm/src/device_manager.rs:2041 -- Using asynchronous RAW disk file (io_uring)
cloud-hypervisor: 15.336665ms: <vmm> INFO:virtio-devices/src/block.rs:445 -- Disk topology: DiskTopology { logical_block_size: 512, physical_block_size: 512, minimum_io_size: 512, optimal_io_size: 0 }
cloud-hypervisor: 15.4434ms: <vmm> INFO:vmm/src/device_manager.rs:2274 -- Creating virtio-rng device: RngConfig { src: "/dev/urandom", iommu: false }
cloud-hypervisor: 16.009421ms: <vmm> INFO:vmm/src/vm.rs:2024 -- Booting VM
cloud-hypervisor: 16.121853ms: <vmm> INFO:vmm/src/cpu.rs:707 -- Request to create new vCPUs: desired = 14, max = 14, allocated = 0, present = 0
cloud-hypervisor: 16.181911ms: <vmm> INFO:vmm/src/cpu.rs:673 -- Creating vCPU: cpu_id = 0
cloud-hypervisor: 16.411887ms: <vmm> INFO:vmm/src/cpu.rs:297 -- Configuring vCPU: cpu_id = 0
cloud-hypervisor: 16.503299ms: <vmm> INFO:vmm/src/cpu.rs:673 -- Creating vCPU: cpu_id = 1
cloud-hypervisor: 16.669565ms: <vmm> INFO:vmm/src/cpu.rs:297 -- Configuring vCPU: cpu_id = 1
cloud-hypervisor: 16.731235ms: <vmm> INFO:vmm/src/cpu.rs:673 -- Creating vCPU: cpu_id = 2
cloud-hypervisor: 16.899037ms: <vmm> INFO:vmm/src/cpu.rs:297 -- Configuring vCPU: cpu_id = 2
cloud-hypervisor: 16.961675ms: <vmm> INFO:vmm/src/cpu.rs:673 -- Creating vCPU: cpu_id = 3
cloud-hypervisor: 17.132989ms: <vmm> INFO:vmm/src/cpu.rs:297 -- Configuring vCPU: cpu_id = 3
cloud-hypervisor: 17.210538ms: <vmm> INFO:vmm/src/cpu.rs:673 -- Creating vCPU: cpu_id = 4
cloud-hypervisor: 17.358418ms: <vmm> INFO:vmm/src/cpu.rs:297 -- Configuring vCPU: cpu_id = 4
cloud-hypervisor: 17.42129ms: <vmm> INFO:vmm/src/cpu.rs:673 -- Creating vCPU: cpu_id = 5
cloud-hypervisor: 17.56867ms: <vmm> INFO:vmm/src/cpu.rs:297 -- Configuring vCPU: cpu_id = 5
cloud-hypervisor: 17.627911ms: <vmm> INFO:vmm/src/cpu.rs:673 -- Creating vCPU: cpu_id = 6
cloud-hypervisor: 17.7789ms: <vmm> INFO:vmm/src/cpu.rs:297 -- Configuring vCPU: cpu_id = 6
cloud-hypervisor: 17.838235ms: <vmm> INFO:vmm/src/cpu.rs:673 -- Creating vCPU: cpu_id = 7
cloud-hypervisor: 17.989046ms: <vmm> INFO:vmm/src/cpu.rs:297 -- Configuring vCPU: cpu_id = 7
cloud-hypervisor: 18.055461ms: <vmm> INFO:vmm/src/cpu.rs:673 -- Creating vCPU: cpu_id = 8
cloud-hypervisor: 18.200898ms: <vmm> INFO:vmm/src/cpu.rs:297 -- Configuring vCPU: cpu_id = 8
cloud-hypervisor: 18.261859ms: <vmm> INFO:vmm/src/cpu.rs:673 -- Creating vCPU: cpu_id = 9
cloud-hypervisor: 18.441465ms: <vmm> INFO:vmm/src/cpu.rs:297 -- Configuring vCPU: cpu_id = 9
cloud-hypervisor: 18.500762ms: <vmm> INFO:vmm/src/cpu.rs:673 -- Creating vCPU: cpu_id = 10
[... cut ...] 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 20, 0, 4, 8, b0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 8, 0, 1, c0, 3, 0, 0, 0, 0, 0, 0, 1, 8, 0, 1, c0, 3, 0, 0, 0, 0, 0, 0, f0, d2, d2, 85, c5, 7f, 0, 0]
Error booting VM: VmBoot(PopulateHob(GuestMemoryWriteHob(InvalidGuestAddress(GuestAddress(4278464512)))))
```

</details>

jyao1, 2022-03-17:
It seems misunderstand. We did test for multiple CPUs, after we are in TdShim. But this log shows we are not in TdShim yet.

It seems an unnecessary design to add vCPU to DSDT, because the TDX architecture does not support CPU HotPlug at all.

All CPUs must be present at TD guest entrypoint.

I think the fix should be: removing vCPU entry in DSDT.

sboeuf, 2022-03-17:
@rbradford took care of it with https://github.com/cloud-hypervisor/cloud-hypervisor/pull/3860

rbradford, 2022-03-17:
> @rbradford took care of it with [cloud-hypervisor/cloud-hypervisor#3860](https://github.com/cloud-hypervisor/cloud-hypervisor/pull/3860)

Not at all. The DSDT space being offered by the firmware is simply too small to accommodate all the functionality offered. Removing the CPU and memory hotplug functionality will make it slightly smaller, but more importantly they should be denied on TDX anyway (that was my main motivation for the PR.)

AFAICT there is no reason that PCI device hotplug (of virtio devices) cannot be supported on TDX and the support for that makes up the bulk of the DSDT.

jyao1, 2022-03-17:
NOTE: I did not comment anything for PCI hotplug or Memory HotPlug. They can be supported.

My comment is about CPU hotplug - TDX does not support CPU hotplug.

rbradford, 2022-03-17:
> NOTE: I did not comment anything for PCI hotplug or Memory HotPlug. They can be supported.
> 
> My comment is about CPU hotplug - TDX does not support CPU hotplug.

https://lkml.org/lkml/2022/2/27/433 indicates CPU and memory hotplug are not supported.

Since as you confirm PCI device hotplug is supported the space available in the DSDT must be larger (since PCI device hotplug functionality this makes up most of the size of the DSDT)

rbradford, 2022-03-21:
I ran a test where I generated the largest potential valid ACPI table config using Cloud Hypervisor:

```
target/debug/cloud-hypervisor --kernel ~/src/linux/vmlinux --cmdline "console=hvc0 root=/dev/vda1" --disk path=~/workloads/focal-server-cloudimg-amd64-custom-20210609-0.raw -v --api-socket /tmp/api --memory size=1G --platform num_pci_segments=16 --cpus boot=254
```

And the generated ACPI table total was 
```
cloud-hypervisor: 113.900784ms: <vmm> INFO:vmm/src/acpi.rs:773 -- Generated ACPI tables: took 34690µs size = 84438
```

So I think if you make the space for the table 96KiB I think you should have enough space.

sboeuf, 2022-03-23:
@jyao1 @gaojiaqi7 Do you think you can update the HOB size accordingly?

jiangliu, 2022-03-23:
We are facing some issue related to runtime layout too, still struggling to solve it. The good news is that build time/runtime layout is just a recommendation, we could easily change it.

jyao1, 2022-03-24:
Yes. Agree. We need move the TD_HOB to runtime memory space, and enlarge it.

sboeuf, 2022-03-24:
@jiangliu @jyao1 should we open a different issue to detail what needs to happen, or is this one enough?

gaojiaqi7, 2022-04-08:
TD HOB is moved to low memory and enlarged to 128K bytes, this issue has been fixed: https://github.com/confidential-containers/td-shim/commit/b7393d8969f92dc49de773a72f0dfe9dddb81299

closed 2022-04-08 by gaojiaqi7

sboeuf, 2022-04-13:
@gaojiaqi7 this is great news :)
Have you got a chance to try the latest tdshim version with Cloud Hypervisor?