confidential-containers/td-shim #172: [Bug] Normal IO write will trigger exception 'Protection fault'

liuw1, 2022-03-23:
When using below code to write RTC register in td test payload, it will trigger  exception 'Protection fault'.
```unsafe { x86::io::outb(0x70, 0x32) };```

[image]

liuw1, 2022-04-26:
Jiaqi will take this issue

liuw1, 2022-04-30:
#253 only solved io write issue, but io read still have issue as below:

```
INFO - [Test: testve]
INFO - Double fault
INFO - RFLAG: 0000000000010002
INFO - CS:    0000000000000038
INFO - RIP:   000000000121F6C6
INFO - CODE:  0000000000000000
INFO - RAX:   0000000001236D32
INFO - RCX:   000000027F000108
INFO - RDX:   000000027F000071
INFO - RDI:   000000027F000128
INFO - RSI:   000000027F000108
INFO - R8:    0000000000000D20
INFO - R9:    0000000000000010
INFO - R10:   00000000012398E8
INFO - R11:   0000000000000040
INFO - RBX:   0000000000000070
INFO - RBP:   00000000012384C8
INFO - R12:   0000000001238384
INFO - R13:   000000007F7FCC18
INFO - R14:   0000000001238370
INFO - R15:   000000007F7FCC98
```

gaojiaqi7, 2022-04-30:
Ok it seems there's still some mistakes inside VE handler. I've tried to trigger INT n then VE it looks good, but if I trigger VE first then another exception will cause the double fault...

gaojiaqi7, 2022-05-03:
@liuw1 Can you try with https://github.com/confidential-containers/td-shim/pull/257, it should have fixed #DF

liuw1, 2022-05-04:
@gaojiaqi7  The test case of VE pass with #257

jyao1, 2022-05-16:
done

jyao1, 2022-05-16:
user mistake, @257 is not merged yet.

gaojiaqi7, 2022-05-19:
this issue has been fixed by: https://github.com/confidential-containers/td-shim/commit/e84430c85154feb13301a49f9a23d3341ff314c5

closed 2022-05-19 by gaojiaqi7