confidential-containers/td-shim #173: Consider doing periodic releases and storing the release artifacts as part of the GitHub repo.

fidencio, 2022-03-23:
For projects consuming td-shim, it'd be very nice if we could do releases on the staging branch and let other projects direct consume the binaries built from there.

This would help a ton with what Kata Containers will end up shipping as part of their artefacts, and also on how TDX tests will be run as part of the confidential containers effort.

For the Cloud Hypervisor integration, we'd need a binary td-shim built with `boot-kernel` support, for instance.

For now this is just a way to gather feedback on the idea on having it done here.

jyao1, 2022-03-25:
@jiangliu, @sameo Any idea on release tag on staging? I am OK on that proposal.

Our current priority is to move modules from staging to main trunk...

fidencio, 2022-03-30:
@jyao1, @jiangliu, @sameo,

What about starting with 0.0.1-alpha0, 0.0.1-alpha1, ..., 0.0.1-alphaN ... weekly releases on the staging branch, and once everything is on `main` the first 0.0.1 (or even 1.0.0?) release is cut?

At this point, in order to ease the tests of Kata Containers, using Cloud Hypervisor with td-shim, any release of td-shim built with the  `boot-kernel` feature would be very much welcome.

sameo, 2022-04-07:
I would not even bother with the `-alpha` suffixes. From `staging` I would go with `v0.0.N` releases every week and then move to `0.M.N` where `N>0` once we move to main. A `td-shim` `v0.0.1` or `v0.0.2` release is self-explanatory imho.

And `staging` is just a branch name, we can and should start doing releases from it, I agree.

fidencio, 2022-04-13:
@jyao1, @jiangliu, polite ping.

Can we move towards this?  What's needed to move toward this direction?

fidencio, 2022-04-13:
I just synced with @jyao1 on this, internally, and we agreed on the following:
* We have a release tagged every now and then;
* This release will be smoke tested from my side, to ensure the binary works well with cloud-hypervisor and cloud-hypervisor + kata-containers;
* ideally we'd have the release process automated via GitHub actions (which I will provide);

We'll work on this list from the last item to the first.

jyao1, 2022-04-13:
Thanks @fidencio .