confidential-containers/td-shim #255: Need resolve fossa license failure

jyao1, 2022-05-01:
https://app.fossa.com/projects/git%2Bgithub.com%2Fconfidential-containers%2Ftd-shim?ref=badge_shield

dcmiddle, 2022-05-12:
FYI per @magowan https://github.com/confidential-containers/community/wiki/CNCF-Fossa

peterzcst, 2022-05-12:
A summary about the FOSSA issues below in case above link is complex:
1.	afl (latest) used by:
•	bootloader (0.10.12)
bootloader -> noto-sans-mono-bitmap -> … -> cpp_demangle -> afl
bootloader- > noto-sans-mono-bitmap -> … -> cpp_demangle -> afl
bootloader -> noto-sans-mono-bitmap -> … -> cpp_demangle -> afl
•	clap (3.1.14)
clap -> criterion -> … -> cpp_demangle -> afl
clap -> criterion-> … -> cpp_demangle -> afl
clap -> criterion-> … -> cpp_demangle -> afl
clap -> textwrap-> … -> cpp_demangle -> afl
•	chrono (0.4.19)
chrono -> criterion -> … -> cpp_demangle ->afl
chrono -> criterion -> … -> cpp_demangle ->afl
chrono -> criterion -> … -> cpp_demangle ->afl
 
2.	freetype-sys used by:
•	zerocopy (0.6.1)
zerocopy -> zerocopy-derive -> syn -> … -> servo-fontconfig-sys -> freetype-sys
zerocopy -> zerocopy-derive -> syn ->… -> servo-fontconfig-sys -> freetype-sys
•	bitflags (1.3.2)
Bitflags -> serde_derive -> syn -> -> servo-fontconfig-sys -> freetype-sys
•	clap (3.1.14)
clap -> criterion -> smol -> -> … -> servo-fontconfig-sys -> freetype-sys
clap -> criterion -> smol -> … -> servo-fontconfig-sys -> freetype-sys
clap -> criterion -> async-std -> … -> servo-fontconfig-sys -> freetype-sys
clap -> textwrap -> criterion -> smol -> … ->servo-fontconfig-sys -> freetype-sys
•	chrono (0.4.19)
chrono -> criterion -> smol -> -> … -> servo-fontconfig-sys ->freetype-sys
chrono -> criterion -> smol -> -> … -> servo-fontconfig-sys ->freetype-sys
chrono -> criterion -> async-std -> … -> servo-fontconfig-sys ->freetype-sys
 
3.	gmp-mpfr-sys used by: 
•	x86 (0.47.0)
x86 -> phf_codegen -> phf_generator -> criterion -> async-std -> … -> hdrhistogram -> rug -> gmp-mpfr-sys
x86 -> phf_codegen -> phf_generator -> criterion -> async-std -> … -> hdrhistogram -> rug -> gmp-mpfr-sys
•	clap (3.1.14)
clap -> criterion -> smol -> async-std -> … ->hdrhistogram -> rug -> gmp-mpfr-sys
clap -> criterion -> smol -> async-std -> … ->hdrhistogram -> rug -> gmp-mpfr-sys
clap -> criterion -> smol -> async-std -> … ->hdrhistogram -> rug -> gmp-mpfr-sys
clap -> textwrap-> criterion -> smol -> async-std -> … ->hdrhistogram -> rug -> gmp-mpfr-sys
clap -> trycmd -> toml_edit -> criterion -> smol -> surf -> tide-> hdrhistogram ->rug -> gmp-mpfr-sys
•	chrono (0.4.19)
chrono -> criterion -> smol -> async-std -> … ->hdrhistogram -> rug -> gmp-mpfr-sys
chrono -> criterion -> smol -> async-std -> … ->hdrhistogram -> rug -> gmp-mpfr-sys
chrono -> criterion -> smol -> async-std -> … ->hdrhistogram -> rug -> gmp-mpfr-sys
 
5.	lz4-sys used by:
•	bootloader (0.10.12)
bootloader -> noto-sans-mono-bitmap -> … -> lz4 -> lz4-sys
bootloader -> noto-sans-mono-bitmap -> … -> lz4 ->lz4-sys
 
6.	lzma-sys used by:
•	der (0.4.5)
der -> crypto-bigint -> proptest -> x86 -> raw-cpuid -> clap -> trycmd -> toml_edit -> … -> async-compression -> xz2 -> lzma-sys
dercrypto-bigint -> proptest -> x86 -> raw-cpuid -> clap -> trycmd -> toml_edit -> … -> async-compression -> xz2 -> lzma-sys
•	x86 (0.47.0)
x86 -> raw-cpuid -> clap -> trycmd -> toml_edit -> … -> async-compression -> xz2 -> lzma-sys
x86 -> klogger -> x86 -> raw-cpuid -> clap -> trycmd -> toml_edit -> … -> async-compression -> xz2 -> lzma-sys
•	clap (3.1.14)
clap -> criterion -> tokio -> proptest -> x86 -> raw-cpuid -> clap -> trycmd -> toml_edit -> … -> async-compression -> xz2 -> lzma-sys
clap -> trycmd -> toml_edit -> … -> async-compression -> xz2 -> lzma-sys
clap -> textwrap -> criterion -> tokio -> proptest -> x86 -> raw-cpuid -> clap -> trycmd -> toml_edit -> … -> async-compression -> xz2 ->lzma-sys
•	chrono (0.4.19)
chrono ->criterion -> tokio -> proptest -> x86 -> raw-cpuid -> clap -> trycmd -> toml_edit -> … -> async-compression -> xz2 -> lzma-sys
chrono -> criterion -> tokio -> proptest -> x86 -> raw-cpuid -> clap -> trycmd -> toml_edit -> … -> async-compression -> xz2 -> lzma-sys
•	bootloader (0.10.12)
bootloader -> raw-cpuid -> clap -> trycmd -> toml_edit -> … -> async-compression -> xz2 -> lzma-sys
 
7.	rug used by: 
[... 26 lines cut ...]
bitflags -> serde_derive -> syn -> reqwest -> native-tls -> security-framework -> x509-parser -> der-parser -> … -> servo-fontconfig -> servo-fontconfig-sys -> servo-freetype-sys
•	[clap (3.1.14)](https://app.fossa.com/projects/git%2Bgithub.com%2Fconfidential-containers%2Ftd-shim/refs/branch/master/7d0aeae4eed6e60dcd32d9af06ad97ab67896cf2/browse/dependencies/cargo%2Bclap%243.1.14)
clap -> criterion -> smol -> tide -> surf -> http-client -> hyper -> … -> servo-fontconfig -> servo-fontconfig-sys -> servo-freetype-sys
clap -> criterion -> smol -> surf -> http-client -> tide -> http-client -> hyper -> … -> servo-fontconfig -> servo-fontconfig-sys -> servo-freetype-sys
clap -> criterion -> async-std -> surf -> http-client -> tide -> http-client -> hyper -> … -> servo-fontconfig -> servo-fontconfig-sys -> servo-freetype-sys
clap -> textwrap -> criterion -> smol -> tide -> surf -> http-client -> hyper -> … -> servo-fontconfig -> servo-fontconfig-sys -> servo-freetype-sys
•	chrono (0.4.19)
chrono -> criterion -> smol -> surf -> http-client -> hyper -> … -> servo-fontconfig -> servo-fontconfig-sys -> servo-freetype-sys
chrono -> criterion -> async-std -> surf -> http-client -> tide -> http-client -> hyper -> … -> servo-fontconfig -> servo-fontconfig-sys -> servo-freetype-sys

10.	Cortex-m-semihosting(0.3.7): Not a problem
//! On the host side:
//!
//! ``` text
//! $ openocd -f $INTERFACE -f $TARGET -l /tmp/openocd.log
//! Open On-Chip Debugger 0.9.0 (2016-04-27-23:18)
//! Licensed under GNU GPL v2
//! For bug reports, read
//!         http://openocd.org/doc/doxygen/bugs.html
//! # the command will block at this point
//! ```
•	bootloader (0.10.12)
•	log (0.4.16)
•	clap (3.1.14)
•	chrono (0.4.19)

jyao1, 2022-05-12:
As summary, we need remove bootloader, clap, chrono, x86, der, zerocopy, bitflags.

gaojiaqi7, 2022-05-13:
Look at the deep end dependencies marked by fossa, none of their licenses is GPL, they are marked because 'GPL' words were scaned out in some of their files or submodules.
Should we simply conclude that there are problems with their license and remove the crates relying on them?

jyao1, 2022-05-13:
Below:
* [afl](https://crates.io/crates/afl) - Apache-2.0
* [freetype-sys](https://crates.io/crates/freetype-sys) - MIT
* [gmp-mpfr-sys](https://crates.io/crates/gmp-mpfr-sys) - LGPL-3.0+
* [lz4-sys](https://crates.io/crates/lz4-sys) - MIT
* [lzma-sys](https://crates.io/crates/lzma-sys) - MIT OR Apache-2.0
* [rug](https://crates.io/crates/rug) - LGPL-3.0+
* [sdl2-sys](https://crates.io/crates/sdl2-sys) - MIT
* [servo-freetype-sys](https://crates.io/crates/servo-freetype-sys) - FTL OR GPL-2.0

I think we need remove LGPL.
For rest, should we check with fossa owner?

gaojiaqi7, 2022-05-13:
But these crates do not appear in the Cargo.lock file of td-shim, so it seems that td-shim does not use code of these crates directly

fitzthum, 2022-05-18:
There is a little bit of information about licenses for CNCF [here](https://github.com/cncf/foundation/blob/main/allowed-third-party-license-policy.md). Note that MIT licenses are pre-approved by the CNCF (as long as you don't modify the source and the project wasn't created yesterday). Apache-2.0 projects are allowed by default. I'm not exactly sure why either of those licenses are flagged by FOSSA.

We probably do want to remove GPL/LPL/FTL stuff at some point.

ariel-adam, 2022-10-11:
@jyao1 is this issue still relevant or can be closed? 
If it's still relevant to what release do you think we should map it to (mid-November, end-December, mid-February etc...)?

jyao1, 2022-12-17:
no issue found now.

closed 2022-12-17 by jyao1