confidential-containers/td-shim #256: [Bug] td-shim should support reproducible build

liuw1, 2022-05-02:
There are some differences between 2 times build with same code base. Currently td-shim does not support reproducible build.

[image]

jiangliu, 2022-05-02:
It's really import to get this fixed, reproducible build is fatal for CC:)
Thanks for working this!

jyao1, 2022-05-12:
Some investigation result from @longlongyang.

Elements break reproducible build (x86_64-unknown-uefi + target x86_64-unknown-none)
1.	Debug related sections (PE + ELF)
2.	Code base path like \home\long\rust-tpa\xxx\xxx.rs (PE + ELF)
3.	CARGO_HOME like \home\long\.cargo\xxx\xxx.rs (PE + ELF)
4.	RUSTUP_HOME like \home\long\.rustup\xxx\xxx.rs (PE + ELF)
5.	TimeDateStamp field in COFF file header (PE only)

Solution:
1.	Use `strip = "symbols"` option to `[profile.release]`
2.	Use `strip = "symbols"` option to `[profile.release]`
3.	Use a post-build tool to zero CARGO_HOME string
4.	Use a post-build tool to zero RUSTUP_HOME string
5.	Use a post-build tool to zero TimeDateStamp for PE

NOTE: 3 and 4 may be resolved directly after rust-lang RFC 3127. It is temp solution before RFC is implemented.

longlongyang, 2022-05-12:
I would like to take this issue.

jyao1, 2022-05-20:
After more investigation, we notice that Use strip = "symbols" option to [profile.release] is not stable.

It is just added in latest 1.59 version - https://github.com/confidential-containers/td-shim/issues/269.

And it causes weird issue - https://github.com/confidential-containers/td-shim/issues/272

As such, I think we need use post-build tool to resolve 1 and 2 as well.

After the tool is stable, we can use strip again.

PR #268 (2022-05-27): Reproducible build: introduce a tool to help reproducible build
fix #256

1. Introduce a reproducible tool, td-reproducible-tool, to help reproducible
2. Add strip = "symbols" to cargo.toml in root workspace to remove symbols
closed 2022-05-27 by jyao1