confidential-containers/td-shim #269: Need to upgrade the toolchain to 1.59 release to accept "strip" option in release profile

longlongyang, 2022-05-16:
Background: https://github.com/confidential-containers/td-shim/issues/256#issuecomment-1124460655
The current stable toolchain version 1.58 release doesn't support the "strip" option in release profile, which breaks the reproducible build. 
We need to upgrade the version of toolchain to 1.59 release, since the profile strip option has been stabilized in the 1.59 release. More detail please find: https://doc.rust-lang.org/cargo/reference/unstable.html#profile-strip-option

jyao1, 2022-05-16:
@liuw1, can you take a look to upgrade the version in CI ?

It causes the CI failure - https://github.com/confidential-containers/td-shim/pull/268

liuw1, 2022-05-16:
Sure

liuw1, 2022-05-16:
I have checked all CI related test tasks except TDX integration test task with #268, all tasks passed with the latest stable toolchain.

longlongyang, 2022-05-16:
> I have checked all CI related test tasks except TDX integration test task with #268, all tasks passed with the latest stable toolchain.

Thanks, I am working on to solve that TDX integration test task failure

PR #270 (2022-05-16): Update stable rust toolchain to 1.60.0
Fix #269
commit 552e40dd (2022-05-16): Update stable rust toolchain to 1.60.0

Fix #269
closed 2022-05-16 by jyao1