confidential-containers/td-shim #272: [bug] secure_boot_verify_payload fails with InvalidSignature when using ECDSA_NIST_P384_SHA384 crypto suite

longlongyang, 2022-05-18:
secure_boot_verify_payload fails with InvalidSignature when using ECDSA_NIST_P384_SHA384 crypto suite with strip = "symbols" set in cargo.toml.
Location:
https://github.com/confidential-containers/td-shim/blob/552e40dd0b5b3dadc3bd9ae30c7b223d26cfe7f6/td-shim/src/bin/td-shim/verifier.rs#L162-L164
How to reproduce the bug:

- If no panic is observed using the code change https://github.com/confidential-containers/td-shim/pull/268, 

  Try add 
  `log::info!("Just visit the image memory, and the bug will be triggered: {:?}\n", self.image[100]);`
  to the line before line 162
  https://github.com/confidential-containers/td-shim/blob/552e40dd0b5b3dadc3bd9ae30c7b223d26cfe7f6/td-shim/src/bin/td-shim/verifier.rs#L162

- if panic is observed using the code change https://github.com/confidential-containers/td-shim/pull/268, then you have reproduced the bug.

Note: The trigger condition of this bug is not fixed. At very first, I caught the bug without visiting the image(message) memory, and the bug is gone after have a visiting in the image memory; And in the other day, I have to visit the image memory in order to trigger the bug.

FYR:
The signer:
https://github.com/confidential-containers/td-shim/blob/552e40dd0b5b3dadc3bd9ae30c7b223d26cfe7f6/td-shim-tools/src/signer.rs#L103-L122
The verifier:
https://github.com/confidential-containers/td-shim/blob/552e40dd0b5b3dadc3bd9ae30c7b223d26cfe7f6/td-shim/src/bin/td-shim/verifier.rs#L55-L89

longlongyang, 2022-05-24:
may be caused by https://github.com/confidential-containers/td-shim/issues/279

ariel-adam, 2022-10-11:
@longlongyang is this issue still relevant or can be closed? 
If it's still relevant to what release do you think we should map it to (mid-November, end-December, mid-February etc...)?

longlongyang, 2022-10-12:
> @longlongyang is this issue still relevant or can be closed? If it's still relevant to what release do you think we should map it to (mid-November, end-December, mid-February etc...)?

Not much relevant I think, This is an unexpected observation when enabling striping symbols. But since we now have a tool to do the striping work, this can be closed , or map it to mid-February to keep track of it

ariel-adam, 2022-10-12:
Similar comment, I will close it for now and you can open when relevant

closed 2022-10-12 by ariel-adam