confidential-containers/td-shim #279: [Bug] Unpredictable behavior may occur if log level features are not specified when enabling strip = "symbol" or "debuginfo" option

longlongyang, 2022-05-23:
The unpredictable behavior includes secure boot test failure, ACPI table not found, E820 table not found, etc. After specifying log level features in dependency "log" crate, The "crash" is not observed any more. 
The possible solution: 
https://github.com/confidential-containers/td-shim/blob/8c2b0ab686fd02db5af954c6a0af02eec1dea580/td-logger/Cargo.toml#L12

gaojiaqi7, 2022-09-08:
The error appears to be due to the use of `xbuild`. If we use `cargo build -p td-shim --target=x86_64-unknown-uefi -Z build-std=core,alloc -Z build-std-features=compiler-builtins-mem --release --features=main,tdx --no-default-features` to build td-shim, this issue will not exist any more.

gaojiaqi7, 2022-09-08:
Similar to the problem with `sha2`.
If we do not force the software implementation (then it will compile the avx2 related code which is not enabled in target `x86_64-unknown-uefi` by default):
`cargo xbuild` will build pass but fail to run.

`cargo build` debug build will fail due to:
```
LLVM ERROR: Do not know how to split the result of this operator!
error: could not compile `sha2`
```
`cargo build` release build will compile and run successfully.

ariel-adam, 2022-10-11:
@longlongyang is this issue still relevant or can be closed? 
If it's still relevant to what release do you think we should map it to (mid-November, end-December, mid-February etc...)?

longlongyang, 2022-10-12:
> @longlongyang is this issue still relevant or can be closed? If it's still relevant to what release do you think we should map it to (mid-November, end-December, mid-February etc...)?

Not much relevant I think, This is an unexpected observation when enabling striping symbols. But since we now have a tool to do the striping work, this can be closed , or map it to mid-February to keep track of it

ariel-adam, 2022-10-12:
I will close it for now and feel free to open if and when relevant (map it to the CoCo releases project and the correct drop)

closed 2022-10-12 by ariel-adam