confidential-containers/td-shim #294: use sha2 to replace ring

jyao1, 2022-05-31:
Current td-shim need use SHA384 to calculate the MR extend.

Since ring upstream is blocked, we should use sha2 to replace ring.

Rust crypto provides https://github.com/RustCrypto/hashes. SHA384 is available in https://crates.io/crates/sha2

gaojiaqi7, 2022-05-31:
We still use ring to verify the payload signature, is there a substitute for this?

jyao1, 2022-05-31:
That is secure boot specific, right? Let's first put ring to secure-boot feature.

jyao1, 2022-09-07:
For secure boot, we may consider https://crates.io/crates/openssl

jodh-intel, 2022-09-07:
Related: https://github.com/confidential-containers/community/issues/44.

ariel-adam, 2022-10-11:
@jyao1 is this issue still relevant or can be closed? 
If it's still relevant to what release do you think we should map it to (mid-November, end-December, mid-February etc...)?

dcmiddle, 2022-10-11:
@jyao1 can you clarify:
> Since ring upstream is blocked, we should use sha2 to replace ring.

Is there a blocked issue or something we can track?

dcmiddle, 2023-03-10:
@jyao1 is this issue resolved now via #428?

closed 2023-03-11 by jyao1