confidential-containers/td-shim #325: [BUG] Can not get payload_info correctly from `read_from_hob`

a74731248, 2022-06-24:
I find that `read_from_hob` can not parser payload_info correctly which was passed from vmm. And I wrote a test function in`td_hob.rs`：
```
#[cfg(test)]
mod test {
    use core::mem::size_of;
    use core::ptr::slice_from_raw_parts;

    use super::TdHobInfo;
    use crate::td_hob::{GuidExtension, Header, HOB_TYPE_END_OF_HOB_LIST, HOB_TYPE_GUID_EXTENSION};
    use td_shim::{PayloadInfo, TD_KERNEL_INFO_HOB_GUID};

    #[test]
    fn test_read_from_hob() {
        let mut buf = [0xaau8; 128];
        let guid = GuidExtension {
            header: Header {
                r#type: HOB_TYPE_GUID_EXTENSION,
                length: size_of::<GuidExtension>() as u16 + size_of::<PayloadInfo>() as u16,
                reserved: 0,
            },
            name: TD_KERNEL_INFO_HOB_GUID.as_bytes().clone(),
        };
        let payload_info = PayloadInfo {
            image_type: 0x2,
            entry_point: 0x2200000,
        };

        let buf1 = unsafe {
            &*slice_from_raw_parts(
                &guid as *const GuidExtension as *const u8,
                size_of::<GuidExtension>(),
            )
        };
        let buf2 = unsafe {
            &*slice_from_raw_parts(
                &payload_info as *const _ as *const u8,
                size_of::<PayloadInfo>(),
            )
        };
        buf[..size_of::<GuidExtension>()].copy_from_slice(buf1);
        buf[size_of::<GuidExtension>()..size_of::<GuidExtension>() + size_of::<PayloadInfo>()]
            .copy_from_slice(buf2);

        let end = Header {
            r#type: HOB_TYPE_END_OF_HOB_LIST,
            length: 0,
            reserved: 0,
        };
        let buf3 = unsafe {
            &*slice_from_raw_parts(&end as *const Header as *const u8, size_of::<Header>())
        };
        buf[size_of::<GuidExtension>() + size_of::<PayloadInfo>()
            ..size_of::<GuidExtension>() + size_of::<PayloadInfo>() + size_of::<Header>()]
            .copy_from_slice(buf3);

        let hob_info = TdHobInfo::read_from_hob(&buf);
        let paylaod_resutl = hob_info.unwrap().payload_info.unwrap();
        assert_eq!(paylaod_resutl.image_type, payload_info.image_type);
        assert_eq!(paylaod_resutl.entry_point, payload_info.entry_point);
    }
} 
```

And tset it in such 2 situation：
1. `PayloadInfo` was design as origin, and test goes fail.
```
pub struct PayloadInfo {
    pub image_type: u32,
    pub entry_point: u64,
} 
```

2.  `PayloadInfo` was design as follow, and test success.
```
pub struct PayloadInfo {
    pub entry_point: u64,
    pub image_type: u32,
} 
```

And I think maybe the problem is [here](https://github.com/confidential-containers/td-shim/blob/d7235f577903766cf60a85c8656b511e0a86aae4/td-shim/src/bin/td-shim/td_hob.rs#L43)

gaojiaqi7, 2022-06-24:
> > 
> 
> I know current layout was defined in the Spec. And I mean, you can try write something to `PayloadInfo` in cloud-hypervisor and pass it to td-shim, and than read this info in td-shim, you will find the payload_info you read in td-shim was not same as what you write in cloud-hypervisor.

Yes you are right :) , cloud-hypervisor writes the structure with padding but 'pread' is reading from slice without padding, I'll fix this

a74731248, 2022-06-26:
> Yes you are right :) , cloud-hypervisor writes the structure with padding but 'pread' is reading from slice without padding, I'll fix this

thanks!

gaojiaqi7, 2022-08-01:
close with a5beed7515ccf600491a3bc1b7ddb46ad7d0eb59

closed 2022-08-01 by gaojiaqi7