confidential-containers/td-shim #349: Trying to understand E820 table vs Resource HOBs

dimakuv, 2022-07-28:
I ran my simple TD payload and collected the info from FV and Resource HOBs, as well as the E820 table. I also analyzed the debug info printed by the TD-Shim itself during startup.

I got these memory ranges:
```
TDSHIM reported Resource HOBs
-----------------------------
System memory : 0x000000000 - 0x0000820000 (PRESENT INITED TESTED ENCRYPTED)   (0MB - 8.125MB)
Reserved      : 0x000820000 - 0x0000840000 (PRESENT INITED TESTED ENCRYPTED)   (8.125MB - 8.25MB)
System memory : 0x000840000 - 0x0080000000 (PRESENT INITED TESTED ENCRYPTED)   (8.25MB - 2048MB)
Reserved      : 0x0feffc000 - 0x00ff000000 (PRESENT INITED TESTED ENCRYPTED)   (4080MB - 4080.25MB)
Reserved      : 0x0ff040000 - 0x00ff041000 (PRESENT INITED TESTED ENCRYPTED)   (4080.25MB - 4080.254MB)
Reserved      : 0x0ff042000 - 0x00ff062000 (PRESENT INITED TESTED ENCRYPTED)   (4080.259MB - 4080.38MB)
Reserved      : 0x0ff062000 - 0x00ff082000 (PRESENT INITED TESTED ENCRYPTED)   (4080.38MB - 4080.51MB)
System memory : 0x100000000 - 0x0880000000 (PRESENT INITED TESTED ENCRYPTED)   (4GB - 34GB)

MMIO          : 0x080000000 - 0x00fee00000 (PRESENT INITED UNCACHEABLE)        (2048MB - 4078MB)
MMIO          : 0x880000000 - 0x1080000000 (PRESENT INITED UNCACHEABLE)        (34GB - 66GB)

Received Resource HOBs
----------------------
System memory : 0x000000000 - 0x0000820000 (PRESENT INITED TESTED ENCRYPTED)   (0MB - 8.125MB)
System memory : 0x000820000 - 0x0000840000 (PRESENT INITED TESTED ENCRYPTED)   (8.125MB - 8.25MB)
System memory : 0x000840000 - 0x0080000000 (PRESENT INITED TESTED ENCRYPTED)   (8.25MB - 2048MB)
System memory : 0x0feffc000 - 0x00ff000000 (PRESENT INITED TESTED ENCRYPTED)   (4080MB - 4080.25MB)
FV hob        : 0x0ff082000 - 0x00ffcaf000 (PRESENT INITED TESTED ENCRYPTED)   (4080.51MB - 4092.7MB)
System memory : 0x100000000 - 0x0880000000 (PRESENT INITED TESTED ENCRYPTED)   (4GB - 34GB)

E820 Table
----------
AddressRangeMemory   : 0x000000000 - 0x000800000 (0MB - 8MB)         // TD-Shim ran here, now usable
AddressRangeReserved : 0x000800000 - 0x000820000 (8MB - 8.125MB)     // Page tables (0x800000 installed into CR3)
AddressRangeMemory   : 0x000820000 - 0x001200000 (8.125MB - 18MB)    // usable
AddressRangeReserved : 0x001200000 - 0x008000000 (18MB - 128MB)      // TD payload image (why so big??)
AddressRangeMemory   : 0x009200000 - 0x07efbe000 (146MB - 1885MB)    // usable
AddressRangeReserved : 0x07efbe000 - 0x07f7be000 (2031MB - 2039MB)   // current stack
AddressRangeMemory   : 0x07f7be000 - 0x07fdfe000 (2039MB - 2046MB)   // TD shim stack (HOBs are there)? now usable
AddressRangeACPI     : 0x07fdfe000 - 0x07fefe000 (2046MB - 2047MB)   // ACPI tables
AddressRangeNVS      : 0x07fefe000 - 0x080000000 (2047MB - 2048MB)   // TD Event Log??
AddressRangeMemory   : 0x0feffc000 - 0x0ff000000 (4079.9MB - 4080MB) // what are these 16KB??
AddressRangeMemory   : 0x100000000 - 0x780000000 (4GB - 34GB)        // usable
```

I have several questions:
1. Why the Resource HOBs received by my TD payload (through the RDI register) do *not* contain MMIO ranges?
2. Why the Resource HOBs received by my TD payload (through the RDI register) do *not* contain Reserved ranges?
3. Why did the range `0x000820000 - 0x0000840000` transform from Reserved to System Memory?
4. Why is the range `0x000800000 - 0x000820000` (currently installed Page Tables) marked as AddressRangeReserved in E820 table, but as System Memory in the Resource HOB?
5. Why is the range `0x001200000 - 0x008000000` (looks like this is the TD Payload image) marked as AddressRangeReserved in E820 table, but is considered a part of System Memory in the Resource HOB?
6. Why is the range `0x07efbe000 - 0x07f7be000` (current stack) marked as AddressRangeReserved in E820 table, but is considered a part of System Memory in the Resource HOB?
6. Same question for the ranges of ACPI tables and the TD Event Log -- why are they marked as System Memory in the Resource HOB?
7. `0x0feffc000 - 0x0ff000000` -- what are these 16KB of memory?
8. Why there are two holes in the E820 table?
    - `0x008000000 - 0x009200000` (`128MB - 146MB`)
    - `0x07efbe000 - 0x07efbe000` (`1885MB - 2031MB`)

I am also looking at the Example Image Layout in https://github.com/confidential-containers/td-shim/blob/main/doc/td-shim-introduction.pdf, and many items on that diagram do *not* relate to what I observe in my TD payload. Is that example stale?

Finally, what is the relationship between Resource HOBs and the E820 table? From what I understand, Resource HOBs are mostly required to identify physical memory ranges (and their characteristics), whereas the E820 table is used by the OS kernel to decide on the actual ranges to be used. But then why the Page Tables range is marked as AddressRangeReserved -- the OS kernel can create its own Page Tables and then re-use the old ones as normal free memory.

Sorry for the long thread. My main question is: **Should I rely on the Resource HOBs or the E820 table in my OS kernel?**. In other words, **when should I use Resource HOBs** and **when should I use the E820 table**?

dimakuv, 2022-07-28:
I guess I found the answer to one of my questions.

> 1. Why the Resource HOBs received by my TD payload (through the RDI register) do not contain MMIO ranges?

This is because, according to https://uefi.org/specs/ACPI/6.4/15_System_Address_Map_Interfaces/uefi-assumptions-and-limitations.html: `The firmware does not return a range description for the memory mapping of PCI devices, ISA Option ROMs, and ISA Plug and Play cards because the OS has mechanisms available to detect them.`

gaojiaqi7, 2022-07-28:
Thanks @dimakuv for you feedback. According to the TD Shim/Payload interface defined here: https://github.com/confidential-containers/td-shim/blob/main/doc/tdshim_spec.md#part-iii---td-shim--guest-payload-interface, a payload should rely on the E820 table to get the memory resource information rather than the resource HOBs.

The resource HOBs are still there inside the payload HOB list produced by TD-Shim because we implemented some payloads at earlier time that depend on these, but it should be removed and not be used any more. I will update the implementation in payload_hob.rs to avoid misunderstanding.

Please refer to https://github.com/confidential-containers/td-shim/blob/main/doc/tdshim_spec.md#memory-map for the details about the memory map and MMIO.

jyao1, 2022-07-28:
@dimakuv, would you mind to tell us how you run the system to get the result?

via KVM? via cloud-hypervisor? or something else?

Also which version ? We have seen some inconsistent result before.

gaojiaqi7, 2022-07-28:
> 8. `0x0feffc000 - 0x0ff000000` -- what are these 16KB of memory?
This piece of memory is reserved by QEMU for an EPT identity map and a TSS in order to use vm86 mode to emulate 16-bit code directly (which is used in TDX case) and reported as `reserved` type which is same as the private memory in the old-version QEMU, once we solved issue https://github.com/confidential-containers/td-shim/issues/336, TD-Shim can distinguish this kind of memory from private memory type. We handled this in our memory map test case: https://github.com/confidential-containers/td-shim/blob/main/tests/test-td-payload/src/testmemmap.rs#L109-#L118

> 9. Why there are two holes in the E820 table?
>    
>    * `0x008000000 - 0x009200000` (`128MB - 146MB`)
>    * `0x07efbe000 - 0x07efbe000` (`1885MB - 2031MB`)
>
I think the first range `0x001200000 - 0x009200000` rather than `0x001200000 - 0x8000000` which is payload memory,
and the second '0x7EFBE000 - 0x7F7BE000' which is payload stack

For the big payload memory, it can be configured with the https://github.com/confidential-containers/td-shim/tree/main/devtools/td-layout-config, it can automatically generate `runtime.rs` based on JSON config file. The default configuration is used to handle vmlinux case. And we'll do some optimization to reserve the `real` memory usage of loading payload instead of whole piece of memory in the future.

dimakuv, 2022-07-29:
@jyao1 For my environment, please see my other answer: https://github.com/confidential-containers/td-shim/issues/350#issuecomment-1199022290

@gaojiaqi7 Thank you very much for the detailed answers! I should have looked at the https://github.com/confidential-containers/td-shim/blob/main/devtools/td-layout-config/config.json file before asking my question. It has a lot of helpful information.

dimakuv, 2022-07-29:
@gaojiaqi7 @jyao1 Sorry but I have two more questions.

1. Do I understand correctly that this range: `AddressRangeReserved : 0x000800000 - 0x000820000 (8MB - 8.125MB)     // Page tables` can be actually used by the OS kernel as soon as it creates its own page tables and rewires `CR3` to point to the new page tables?

2. How can the layout be changed? I observe that it's not enough to modify `devtools/td-layout-config/config.json` but one should also modify `td-layout/src/runtime.rs` and then rebuild. I was considering changing the layout for my purposes (for simplicity of my TD Shim), but it looks like this effort is non-trivial and not well-supported by current TD Shim (it needs a rebuild and may need some other code changes).

So do I understand correctly that the layout of TD Shim should *not* be modified by users? Or maybe you have a plan to allow users/TD payload developers to specify their own layouts of the final TD-shim blob? Is this layout subject to change?

It seems that the memory layout is subject to change, and it happens pretty frequently, and that layout modifications sometimes require code changes. I base these assumptions on the commits that I glanced over:
- https://github.com/confidential-containers/td-shim/commit/9b0c4358061d23663eba3fde0740c5121b2aa79d
- https://github.com/confidential-containers/td-shim/commit/1a98305ebae41806889f28ccbb122865f9420bcb

gaojiaqi7, 2022-08-31:
Hi @dimakuv , sorry I missed this.

>     1. Do I understand correctly that this range: `AddressRangeReserved : 0x000800000 - 0x000820000 (8MB - 8.125MB)     // Page tables` can be actually used by the OS kernel as soon as it creates its own page tables and rewires `CR3` to point to the new page tables?
> 
Yes, you are right. Once the OS kernel set its own page table the one reserved by td-shim can be freed. But I think it's not easy to identify which piece of memory is used by page table, because its type in the E820 table is reserved.

> 2. How can the layout be changed? I observe that it's not enough to modify `devtools/td-layout-config/config.json` but one should also modify `td-layout/src/runtime.rs` and then rebuild. I was [...]
You can update the `td-layout/src/runtime.rs` by running the tool under devtools/td-layout-config with `cargo run -- --config config.json ../../td-layout/src/`. TD-Shim repository contains a default configuration for the layout which are the `devtools/td-layout-config/config.json` correspond with the `td-layout/src/runtime.rs`. This is a little confusing and maybe we can use a config file directly in a build process in the future which could be more simple.

And you can have a look at the latest memory map which is cleaner. https://github.com/confidential-containers/td-shim/pull/369.

dimakuv, 2022-08-31:
Thanks, I got all the answers I wanted. I will close this issue now.

closed 2022-08-31 by dimakuv