confidential-containers/td-shim #350: Trying to understand ACPI Table Extension HOB

dimakuv, 2022-07-28:
I ran my simple TD payload and collected the info from ACPI Table Extension HOBs.

TD-Shim reports only two HOBs for me:
1. `APIC` (MADT) table, for local and I/O APICs
2. `CCEL` (TDX-defined) table, for trusted boot

Why I don't see any other ACPI tables? There are no RSDT and XSDT (but these two are probably not needed anyway, since all ACPI tables can be found anyway by iterating over ACPI HOBs).

But there is also no SPRT and FADT.

I'm especially interested in SPRT -- is this something still missing from the TD-Shim implementation?

jyao1, 2022-07-28:
td-shim can only produce what it knows.

FADT is to report fixed hardware. Only VMM knows the emulated resource.
SPRT is for interrupt routing. Only VMM knows the emulated hardware.

SPRT and FADT are unknown by td-shim. They should be input by the VMM, passed from VMM to td-shim, then passed from td-shim to guest kernel.

jyao1, 2022-07-28:
@dimakuv, would you mind to tell us who you run the system to get the result?

via KVM? via cloud-hypervisor? or something else?

dimakuv, 2022-07-29:
@jyao1 I use QEMU/KVM. Here is the info you requested:

```
$ /usr/libexec/qemu-kvm --version
QEMU emulator version 6.1.50 (<I hid the exact version, not sure it can be shown publicly>)
```

TD shim version:
```
commit 5f62a0e367b1845a54e534d103ed4a697a599ac3 (HEAD -> main, origin/main, origin/HEAD)
Date:   Wed Jul 13 14:43:32 2022 +0800
    td-layout: Unit test enhancement
```

The way I run it. First I link my TD payload with the TD-Shim binary:
```
cargo run -p td-shim-tools --features="linker" --no-default-features --bin td-shim-ld -- \
    target/x86_64-unknown-uefi/release/ResetVector.bin \
    target/x86_64-unknown-uefi/release/td-shim.efi \
    -p <my TD payload> -o target/x86_64-unknown-uefi/release/final-elf.bin
```

Then I run the resulting blob like this:
```
/usr/libexec/qemu-kvm -accel kvm -smp 8,sockets=1 -m 32G \
    -vga none -monitor pty -no-hpet -nodefaults \
    -device loader,file=target/x86_64-unknown-uefi/release/final-elf.bin,id=fd0 \
    -object tdx-guest,id=tdx -cpu host,-kvm-steal-time,pmu=off,-pks \
    -machine q35,pic=no,kernel_irqchip=split,kvm-type=tdx,confidential-guest-support=tdx \
    -device virtio-serial,romfile= -chardev stdio,id=mux,mux=on,logfile=vm.log \
    -device virtconsole,chardev=mux -monitor chardev:mux -serial chardev:mux
```

> SPRT and FADT are unknown by td-shim. They should be input by the VMM, passed from VMM to td-shim, then passed from td-shim to guest kernel.

Interesting. This is not how I expected this to work (but I admit, I haven't looked into the VMM -> TD Shim interface in detail).

I thought that QEMU/KVM provides the info to TD Shim in a usual way -- i.e., TD Shim queries the PCI bus, detects the devices, possibly sanitizes them and out of this info generates the SPRT and embeds it into the ACPI Table HOB.

But apparently the logic is different, and the VMM needs to submit the SPRT table to TD Shim (and then TD Shim simply forwards it to the guest kernel).

Which means that I'm using a too-old QEMU/KVM, without the patch for SPRT generation?

jyao1, 2022-07-29:
I would like to clarify something. SPRT is just one way to pass the interrupt info. It is optional. It is just an alternative for DSDT. It is not an requirement for td-shim. A VMM can also pass DSDT to td-shim, to let it pass to guest kernel.

The usage of td-shim is to support simple/lightweight TD environment. Most likely, a confidential container, or a service TD.
If a TD has complex devices and full function OS, such as confidential VM, then TDVF is the solution.

dimakuv, 2022-07-29:
@jyao1 Yes, I understand this.

My initial assumption was that TD Shim realizes that it loads a TD Payload (via `PayloadImageTypeExecutablePayload`) and constructs the SPRT (or prints a warning/fails if SPRT was not provided by the VMM).

But I understand your point now. My assumption was wrong.

So how can I instruct QEMU/KVM to construct an SPRT HOB?

By the way, if Cloud Hypervisor has better support for TD Shim, I can switch to that VMM.

dimakuv, 2022-09-15:
I solved all my problems anyway, so closing this issue.

closed 2022-09-15 by dimakuv