confidential-containers/td-shim #376: [MIRAI] Potential arithmetic overflow in td-shim/memory.rs

liuw1, 2022-08-16:
warning: possible attempt to add with overflow
   --> td-shim/src/bin/td-shim/memory.rs:284:21
    |
284 |                     new.physical_start += to_be_accepted;
    |                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

warning: possible attempt to subtract with overflow
   --> td-shim/src/bin/td-shim/memory.rs:285:21
    |
285 |                     new.resource_length -= to_be_accepted;
    |                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

warning: possible attempt to add with overflow
   --> td-shim/src/bin/td-shim/memory.rs:286:38
    |
286 |                     resources.insert(idx + 1, new);
    |                                      ^^^^^^^

warning: possible attempt to subtract with overflow
   --> td-shim/src/bin/td-shim/memory.rs:298:21
    |
298 |                     to_be_accepted -= size;
    |                     ^^^^^^^^^^^^^^^^^^^^^^

liuw1, 2022-08-16:
Please assign to jiaqi

gaojiaqi7, 2022-08-24:
> warning: possible attempt to add with overflow --> td-shim/src/bin/td-shim/memory.rs:284:21 | 284 | new.physical_start += to_be_accepted; | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>
This can be ensured by checking the validity of the resource hobs

> warning: possible attempt to subtract with overflow --> td-shim/src/bin/td-shim/memory.rs:285:21 | 285 | new.resource_length -= to_be_accepted; | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>
the `to_be_accepted` here is less than or equal to the `resource_length`

> warning: possible attempt to add with overflow --> td-shim/src/bin/td-shim/memory.rs:286:38 | 286 | resources.insert(idx + 1, new); | ^^^^^^^
> 
It is impossible for `idx` to overflow since we don't so much memory to hold the `usize::MAX * size_of::<ResourceDescription>()`

> warning: possible attempt to subtract with overflow --> td-shim/src/bin/td-shim/memory.rs:298:21 | 298 | to_be_accepted -= size; | ^^^^^^^^^^^^^^^^^^^^^^
>
the `size` here is less than or equal to the `to_be_accepted `

liuw1, 2022-09-19:
@gaojiaqi7 
warning: possible attempt to add with overflow --> td-shim/src/bin/td-shim/memory.rs:284:21 | 284 | new.physical_start += to_be_accepted; | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

If the feature is not "lazy-accept", td_be_accepted = u64::MAX. Line 321 may overflow.
https://github.com/confidential-containers/td-shim/blob/553e3e6b3886a56a05decbbab37561da2fc5c534/td-shim/src/bin/td-shim/memory.rs#L311

gaojiaqi7, 2022-09-19:
It should not go into line 321

liuw1, 2022-09-19:
Sorry, I misunderstood it. The fix is OK. So close it.

closed 2022-09-19 by liuw1