confidential-containers/td-shim #377: [MIRAI] Potential arithmetic overflow in td-shim/td/tdx_mailbox.rs

liuw1, 2022-08-16:
warning: possible attempt to add with overflow
   --> td-shim/src/bin/td-shim/td/tdx_mailbox.rs:265:29
    |
265 |             let stack_top = stacks.as_ptr() as u64 + i as u64 * AP_TEMP_STACK_SIZE as u64;
    |                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

warning: possible attempt to add with overflow
   --> td-shim/src/bin/td-shim/td/tdx_mailbox.rs:274:9
    |
274 |         address + align_low + major_part,
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

liuw1, 2022-08-16:
Please assign to jiaqi

gaojiaqi7, 2022-08-24:
> warning: possible attempt to add with overflow --> td-shim/src/bin/td-shim/td/tdx_mailbox.rs:265:29 | 265 | let stack_top = stacks.as_ptr() as u64 + i as u64 * AP_TEMP_STACK_SIZE as u64; | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> 
The `i` here is limited by the `MAX_WORKING_AP_COUNT`, the `stack_top` will not overflow
> warning: possible attempt to add with overflow --> td-shim/src/bin/td-shim/td/tdx_mailbox.rs:274:9 | 274 | address + align_low + major_part, | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>
The `align_low + major_part` is eqaul to the physical resource size which should has been checked before

closed 2022-08-24 by gaojiaqi7