confidential-containers/td-shim #385: Any example/code to mark TD pages as shared?

dimakuv, 2022-08-22:
Is there any API/code in TD-shim that allows to mark TD pages (or whole memory regions) as shared (backed by Shared EPT)?

If there is such code, is there any example of its usage?

I'm trying to understand how one can convert TD-private pages to shared-with-host pages.

gaojiaqi7, 2022-08-23:
hi @dimakuv , there is no such code in td-shim since it doesn't need to communicate with VMM/device using shared page, but you can do it as follows:
 - Get the GPAW from the result of https://github.com/confidential-containers/td-shim/blob/main/tdx-tdcall/src/tdx.rs#L448, and set the shared-bit (GPAW+1, for example, the 52nd bit) of the page table entry of the target page.
 - Flush the TLB.
 - Call https://github.com/confidential-containers/td-shim/blob/main/tdx-tdcall/src/tdx.rs#L264 to map GPA.

Then to convert it back, you can:
 - Clear the shared-bit of the page table entry.
 - Flush the TLB.
 - Call https://github.com/confidential-containers/td-shim/blob/main/tdx-tdcall/src/tdx.rs#L525 to accept memory as private page.

dimakuv, 2022-08-23:
Thanks @gaojiaqi7,

I have some questions.

1. Why do you say that the SHARED bit is `GPAW+1`? I thought it's just `GPAW`. Here is the exerpt from the Intel TDX CPU Architectural Specification:

> a. When 4-level EPT is active, the SHARED bit position would always be bit 47.
> b. When 5-level EPT is active, the SHARED bit position would be bit 47 if GPAW is 0. Otherwise, else it would be bit 51.

2. It looks like VMM can "supress" accesses to shared pages and instead return a #VE exception. Here is the exerpt from the Intel TDX Module Base Architecture Specification:

> For shared memory accesses (when `GPA.SHARED = 1`), as with legacy VMX, the VMM can choose which pages are eligible for #VE mutation based on the value of the Shared EPTE bit 63.

How can I be sure that my TDX-enlightened QEMU/KVM doesn't perform such #VE mutation? Is it controlled by some command-line options to QEMU?

dimakuv, 2022-08-23:
Ok, my question 2 is incorrect. I found the answer in Intel SDM, Volume 3, Section 25.5.7 "Virtualization Exceptions". The #VE exception happens only on specific EPT violations (which shouldn't be the case here, if the pages tables are all correctly set up). That phrase about *pages are eligible for #VE mutation* means that if a corresponding page entry in the EPT has bit 63 set to zero, then specific EPT violations cause #VE in the guest; if the EPT has bit 63 set to one ("Supress #VE"), then specific EPT violations do *not* cause #VE in the guest but instead perform a VM exit to the VMM.

gaojiaqi7, 2022-08-23:
>     1. Why do you say that the SHARED bit is `GPAW+1`? I thought it's just `GPAW`. Here is the exerpt from the Intel TDX CPU Architectural Specification:
right it's a typo, according to the spec: SHARED bit is at GPA bit GPAW - 1. When GPAW is 48, the SHARED bit should be bit 47, and when GPAW is 52, the SHARED bit should be bit 51(from bit 0, the 52nd bit, the last GPA bit)

dimakuv, 2022-08-29:
Thanks again! I'm still stuck with QEMU not recognizing the Shared TD pages, but it's a different problem.

closed 2022-08-29 by dimakuv