confidential-containers/td-shim #402: [MIRAI] Possible physical addresses must not have any bits in the range 52 to 64 set in td-paging

liuw1, 2022-08-31:
warning: possible physical addresses must not have any bits in the range 52 to 64 set
   --> td-paging/src/frame.rs:125:9
    |
125 | /         self.alloc()
126 | |             .map(|addr| PhysFrame::containing_address(PhysAddr::new(addr as u64)))
    | |__________________________________________________________________________________^
    |
note: related location
   --> /home/liuwei/.rustup/toolchains/nightly-2022-08-08-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/option.rs:929:29
    |
929 |             Some(x) => Some(f(x)),
    |                             ^^^^
note: related location
   --> td-paging/src/frame.rs:126:55
    |
126 |             .map(|addr| PhysFrame::containing_address(PhysAddr::new(addr as u64)))
    |                                                       ^^^^^^^^^^^^^^^^^^^^^^^^^^

liuw1, 2022-08-31:
**Reproduce command:**
cd td-paging && MIRAI_FLAGS="--diag=library" cargo mirai

liuw1, 2022-09-21:
@gaojiaqi7 Could you help to take a look at this issue?

liuw1, 2022-10-13:
@ariel-adam Jiaqi has submitted PR to fix this issue. It will be closed when the PR is merged.

ariel-adam, 2022-10-13:
Cool, I've mapped it to the CoCo releases project, drop V0.2.0 priority medium

closed 2022-10-16 by jyao1