confidential-containers/td-shim #418: TD-Shim must warn payload how IDT is set up

dimakuv, 2022-09-15:
I am developing my ELF payload that TD-Shim loads.

## Security/usability issue

My ELF payload performs some `rdmsr`, `wrmsr`, `cpuid` instructions. Under Intel TDX, **only a predefined subset** of MSRs and CPUID leaves can be accessed natively with these instructions; the rest MSRs and CPUID leaves generate a #VE exception, and may be accessed by the #VE handler via `tdvmcall_rdmsr`, `tdvmcall_wrmsr` and `tdvmcall_cpuid`.

In the following, I'll take an example of CPUID. But the same problem relates to MSRs.

TD-Shim installs a #VE exception handler (more specifically, an ISR in the IDT table) during its initialization:  https://github.com/confidential-containers/td-shim/blob/d4cfe2ce56d2052219d3cd7efe878552424c4e21/td-exception/src/idt.rs#L106

This #VE exception handler simply forwards the request from the payload to the host VMM (via `tdvmcall_cpuid`): https://github.com/confidential-containers/td-shim/blob/d4cfe2ce56d2052219d3cd7efe878552424c4e21/td-exception/src/interrupt.rs#L404-L411

There is **no sanitization/verification** of the CPUID leaves. There is also no debug message or anything to inform the developer that a #VE exception was resolved, and the CPUID outputs should be considered **untrusted**.

The TD-Shim specification only says this:
> The expectation for OS is that: OS shall setup its own GDT/IDT/CR3 in the 4KB OS wakeup page immediately and jump to anywhere supported by OS page table.

First, this information is burried under the `Multiple Processor Support` section. Second, this information only refers to the OS kernel, but not to an "arbitrary payload".

I was confused and puzzled why seemingly "not allowed by Intel TDX" accesses to MSRs and CPUID leaves were working... It took me quite some time before I realized that TD-Shim **silently set it up for its own purposes**.

### Proposed solutions

1. Make a clear statement about the #VE exception handling (that it forwards MSR, CPUID, other requests to the host VMM) in the TD-Shim specification. Explain that the results are **untrusted and must be sanitized**.
2. Add a debug message to the #VE exception handler. Explain that the results are **untrusted and must be sanitized**.
3. Somehow force the payload to set up its own GDT/IDT. Well, maybe just spell it out in the TD-Shim specification.

jyao1, 2022-09-15:
Good finding.
1. Agree. We can add more comment there.
2. Disagree. Adding debug message in exception handler is not the best idea.
3. Would you please clarify how to force ? Uninstall #VE before transfer control ?

dimakuv, 2022-09-15:
> 2. Disagree. Adding debug message in exception handler is not the best idea.

True, not the best idea. But TD-Shim could introduce log levels (for debugging purposes), and on the most verbose level, TD-Shim could print some messages even in the exception handler (through TDVMCALL_IO to port 0x3F8). This way I would be able to identify the problem immediately.

> Would you please clarify how to force ? Uninstall #VE before transfer control ?

Yes, this is a good option. Actually, this is what I was expecting when I started writing my payload.

jyao1, 2022-09-15:
I think we can enable 1 and 3 at first.
2 is risky, because of nest interrupt. I think we can revisit after 1 and 3.

closed 2022-12-25 by jyao1