confidential-containers/td-shim #487: [Feature Request] Tools to generate the payload digest

Xynnn007, 2023-01-04:
Hi all, I'm working on the reference values of evidence. Now td-shim calculates the payload's digest via hashing a memory area from `KERNEL_BASE` to `KERNEL_BASE + KERNEL_SIZE` [here](https://github.com/confidential-containers/td-shim/blob/main/td-layout/src/memslice.rs#L82). 

However, when I test with a `bzImage`, the size of the `bzImage` (15791808 bytes) is smaller than `KERNEL_SIZE` (33554432 bytes). The sha384 digest of the `bzImage` also does not match the `Payload` measurement in CCEL. 

Furthermore, I tried to append zeros to the `bzImage` to match the size 33554432 bytes. Still the sha384 digest of the new image does not match. I am not sure what the measured memory interval after the`bzImage` and before `KERNEL_BASE + KERNEL_SIZE` might look like.

I wonder do we have a tool (which I did not find) or do we have a plan to provide a way to calculate a kernel into the expected `Payload` measurement? Thanks!

jyao1, 2023-01-05:
Do you mean MRTD measurement or RTMR measurement?
They use different way.

Xynnn007, 2023-01-05:
@jyao1 I mean RTMR. The code called inside is https://github.com/confidential-containers/td-shim/blob/main/cc-measurement/src/log.rs#L180

jyao1, 2023-01-06:
I think it is possible to add capability to existing https://github.com/confidential-containers/td-shim/tree/main/td-shim-tools/src/bin/td-shim-tee-info-hash.

Xynnn007, 2023-01-06:
I'm not an expert in this. I have two questions:
* For measurement, take `bzImage` for example. Is the whole content of `bzImage` loaded to `KERNEL_BASE` exactly page by page in order?
* You've mentioned that zero padding is involved. To my understanding, pages from `KERNEL_BASE + [len(bzImage) / 4KB]` to `KERNEL_BASE + [KERNEL_SIZE / 4KB]` are pages with all zeros. But what about the last page containing `bzImage`'s data, s.t. the [KERNEL_SIZE / 4KB]-th page? Contents in it are `the end of the bzImage` and `some other data`. I wonder whether the `some other data` are all zeros, or undefined?

> However, when I test with a `bzImage`, the size of the `bzImage` (15791808 bytes) is smaller than `KERNEL_SIZE` (33554432 bytes). The sha384 digest of the `bzImage` also does not match the `Payload` [...]

Here is some tests for the comparation. `17762624 = 33554432 - 15791808`

```bash
[root@localhost release]# ls -al bzImage
-rw-r--r-- 1 root root 15791808 Dec 29 07:23 bzImage
[root@localhost release]# cp bzImage bzImage-extend 
[root@localhost release]# ls -al bzImage-extend 
-rw-r--r-- 1 root root 15791808 Jan  5 21:24 bzImage-extend
[root@localhost release]# dd if=/dev/zero bs=1 count=17762624 >> bzImage-extend
17762624+0 records in
17762624+0 records out
17762624 bytes (18 MB, 17 MiB) copied, 94.7777 s, 187 kB/s
[root@localhost release]# ls -al bzImage-extend 
-rw-r--r-- 1 root root 33554432 Jan  5 21:31 bzImage-extend
[root@localhost release]# sha384sum bzImage-extend 
0f79245eeb87ac333aa7e7e7a20e653387c85a4bcea832210280f201fe3738df0c5ac4bc300981c0934450a8d9789cf5  bzImage-extend
```

The measurement generated by td-shim of this `bzImage` is `5b7aa6572f649714ff00b6a2b9170516a068fd1a0ba72aa8de27574131d454e6396d3bfa1727d9baf421618a942977fa`.

Related test data (CCEL data) is included in this PR https://github.com/confidential-containers/attestation-service/pull/33/files#diff-1a4e5ad4c3b043c019c00bc3b3072fd6e1e5b03a5ce8c498e1c0acaf697d9d3fR231.

We use `cloud-hypervisor` to launch this TEE-VM, and the command is
```bash
./cloud-hypervisor \
    --platform tdx=on \
    --firmware final-boot-kernel.bin\
    --kernel bzImage \
    --cmdline "root=/dev/vda1 console=hvc0 rw" \
    --cpus boot=1 \
    --memory size=1G \
    --net "tap=,ip=,mask=" \
    --log-file  cloudh-debug.log\
    -vvv \
    --serial file=td-shim-to-serial.log \
    --disk path=<path-to-img>
```

Xynnn007, 2023-01-06:
Sorry, there is something wrong with the `KERNEL_SIZE` const. `KERNEL_SIZE` of the fw I used is `268435456` not `33554432`. When I re-padded, it matches.

Xynnn007, 2023-01-09:
I've made a simple tool to calculate this [here](https://github.com/Xynnn007/td-payload-reference-provider), as I'd checked the code in https://github.com/confidential-containers/td-shim/tree/main/td-shim-tools/src/bin/td-shim-tee-info-hash but I didn't know how to contribute to fit in the current CLI design.

Xynnn007, 2023-01-10:
> We can have a standalone tool. Would you please submit https://github.com/Xynnn007/td-payload-reference-provider directly?

Certainly!

closed 2023-01-10 by jyao1