confidential-containers/td-shim #488: Consider use `-Z build-std` instead of `xbuild`

gaojiaqi7, 2023-01-04:
Using `xbuild` requires additional installation. Cargo provides `-Z build-std` to build rust program without standard library.

For example:
```
cargo build -Z build-std=core,alloc,compiler_builtins -p td-shim --target x86_64-unknown-none --release --features=main,tdx
```

jyao1, 2023-01-06:
please evaluate size difference before adopt this change

gaojiaqi7, 2023-02-07:
`-Z build-std` can work with release profile but failed on dev profile (debug build) when lto is enabled, but we have to enable lto otherwise binary will be too large.
```
= note: rust-lld: error: undefined symbol: core::panicking::panic::h218ed11d329799d8
          >>> referenced by x86_64.rs:84 (src/mem/x86_64.rs:84)
          >>>               compiler_builtins-bb203116f4d58bf2.compiler_builtins.878843f6-cgu.6.rcgu.o:(compiler_builtins::mem::memmove::hb0804155d706ca64) in archive /home/jiaqi/source/td-shim/target/x86_64-unknown-none/debug/deps/libcompiler_builtins-bb203116f4d58bf2.rlib
          >>> referenced by x86_64.rs:85 (src/mem/x86_64.rs:85)
          >>>               compiler_builtins-bb203116f4d58bf2.compiler_builtins.878843f6-cgu.6.rcgu.o:(compiler_builtins::mem::memmove::hb0804155d706ca64) in archive /home/jiaqi/source/td-shim/target/x86_64-unknown-none/debug/deps/libcompiler_builtins-bb203116f4d58bf2.rlib
          >>> referenced by x86_64.rs:107 (src/mem/x86_64.rs:107)
          >>>               compiler_builtins-bb203116f4d58bf2.compiler_builtins.878843f6-cgu.6.rcgu.o:(compiler_builtins::mem::memset::h2e2853763d9a5b4a) in archive /home/jiaqi/source/td-shim/target/x86_64-unknown-none/debug/deps/libcompiler_builtins-bb203116f4d58bf2.rlib
          >>> referenced 3 more times
...
```
https://github.com/rust-lang/compiler-builtins/issues/347

closed 2023-02-08 by gaojiaqi7