confidential-containers/td-shim #502: [RFC] td-shim release tagging process

jyao1, 2023-01-18:
Hi I would like to discuss td-shim release tagging process.

**[Background & requirement]**
Ideally, we need to support reproducible build for td-shim.
It means: For a **release** image + same **build environment (compiler + toolchain)**, we can reproduce **same binary** at **any time** and on **any system**.

Reference: https://github.com/confidential-containers/td-shim/issues/256

**[Problem]**
Currently, we do have release tag - which indicates the source code version of td-shim.
But we do not have cargo.lock - that means: Even with same source code, we might not be able to reproduce same binary, because the dependency crate might be upgraded.
That breaks the rule.

Reference: https://doc.rust-lang.org/cargo/guide/cargo-toml-vs-cargo-lock.html

Cargo.lock can be used to lock the version. However, we do want to keep using the latest crate to bug fix, or security fix.
Just put Cargo.lock for the project is not a good idea.

Reference: https://doc.rust-lang.org/cargo/faq.html#why-do-binaries-have-cargolock-in-version-control-but-not-libraries

fidencio, 2023-01-18:
We must consider pinning a version of llvm, llvm-ar, and nasm as well.

One possible way to do so is providing a container image, which could be updated on every change of some parts of the project*, and also provide this container image as part of the tagging process.

Changes to track:
* Cargo.toml
* Cargo.lock
* Toolchain

This process could be fully automated via GitHub actions, as we already do something similar for Kata Containers, and users would be able to use this whenever they "recreate" the build.

jyao1, 2023-01-18:
@fidencio , sounds great. May I assign this to you?

fidencio, 2023-01-20:
> @fidencio , sounds great. May I assign this to you?

I just did, thanks!

dimakuv, 2023-11-08:
@jyao1 @fidencio Could you elaborate on the status of reproducible builds with TD-Shim?

If reproducible builds are implemented, does this apply to TD-Shim + payloads? In the [TDX Migration TD Design Guide document](https://cdrdv2-public.intel.com/733580/tdx-migration-td-design-guide-348987-001.pdf), there is a section on Reproducibility, but it doesn't contain any specific information on whether it is truly supported and how to achieve it exactly (what are the steps to take).

I mention TDX Migration TD simply because IIUC it is TD-Shim + payload, similar to my case.

jyao1, 2023-11-08:
Currently, td-shim supports reproducible build with same user id.

Currently, we use docker - see https://github.com/confidential-containers/td-shim/pull/605.