confidential-containers/td-shim #511: Updating containers started by TD-Shim

ReichertL, 2023-02-06:
Hello, 
I was wondering how containers started by the TD-Shim can be updated.  Can secrets be migrated to a new container? 

There seems to be an option in TDX to include a service TD in the TCB during the start of the TD. The only use case currently defined in the documentation for this is migration. (I.e., the only value for SERVTD_TYPE is Migration TD). 
Do you intend to realize updates through service TDs or maybe by repurposing migration TDs?  If yes, what are the security requirements for updating a container? Same Owner, Same or higher SVN? 

Your help is very much appreciated.

jyao1, 2023-02-08:
Sorry, I don't understand your question.

What kind of "updating" you are referring to?

The generic "updating a container" is irrelevant to TD-Shim, IMHO.

ReichertL, 2023-02-08:
Hello, 
Thank you for the fast reply.

Assume I run a database in a TD container that holds private data. If a vulnerability in the database is found or if I want to add new features, the container would need to be updated. However, I don't want to lose the data in the database.

Intel SGX allowed moving secrets from one enclave to another if they were signed by the same author for updating purposes. The service TDs I mentioned above can potentially be used to realize a similar mechanism. 
Are there plans for such a secret migration mechanism? Or would the owner of a TD container have to download all data (via whatever interface the container software itself provides) and then upload this backup to the updated container?

jyao1, 2023-02-10:
Td-Shim is not the right component to do such thing.

I think it should be done in TD container level.

ReichertL, 2023-02-14:
Thank you for the response, that answers my question.

closed 2023-02-14 by ReichertL