confidential-containers/td-shim #531: Discrepancy of TDVF metadata GUID between upstream EDK II and the MigTD binary

kltseng, 2023-03-01:
HI, I noticed a discrepancy in the GUID between upstream EDK II and the MigTD.

EDK II uses the following byte sequence for the GUID (before the TDVF descriptor) (see link in [1] below):

F3 F9 EA E9 8E 16 D5 44 A8 EB 7F 4D 87 38 F6 AE
==> This encodes the GUID E9EAF9F3-168E-44D5-A8EB-7F4D8738F6AE.

However, in the MigTD file td-shim/td-shim/src/metadata.rs (see link in [2]): 

pub const TDX_METADATA_GUID_STR: &str = "F3F9EAE9-8E16-D544-A8EB-7F4D8738F6AE";
pub const TDX_METADATA_GUID: Guid = Guid::from_fields(
    0xF3F9EAE9,
    0x8E16,
    0xD544,
    [0xA8, 0xEB, 0x7F, 0x4D, 0x87, 0x38, 0xF6, 0xAE],
);

==> The TDX_METADATA_GUID encodes the GUID F3F9EAE9-8E16-D544-A8EB-7F4D8738F6AE (which of course matches the TDX_METADATA_GUID_STR string that it intends to define.)

The first three fields (which are a u32 followed by two u16s) have the byte ordering swapped. Which one is correct?

[1] https://github.com/tianocore/edk2/blob/1eeca0750af5af2f0e78437bf791ac2de74bde74/OvmfPkg/ResetVector/X64/IntelTdxMetadata.asm#L53-L58
[2] https://github.com/confidential-containers/td-shim/blob/13e7035a03d5e30ed189d329a2f828120598d4f5/td-shim/src/metadata.rs#L13

gaojiaqi7, 2023-03-02:
@jyao1 looks like we wanted to use the same GUID as TDVF, but it is not defiend by as specification and used internally. Shall we use the same GUID as TDVF or a completely different one?

jyao1, 2023-03-03:
Please use right order of GUID - `E9EAF9F3-168E-44D5-A8EB-7F4D8738F6AE`

closed 2023-03-07 by jyao1