confidential-containers/td-shim #548: TD Shim panic may cause KVM core dump

liuw1, 2023-03-23:
/usr/libexec/qemu-kvm -accel kvm         -M q35         -name process=tdxvm         -smp 1        -object tdx-guest,id=tdx,debug=off         -machine q35,kernel_irqchip=split,confidential-guest-support=tdx         -no-hpet         -cpu host,host-phys-bits,-kvm-steal-time,pmu=off,-amx-tile,-amx-int8         -bios target/release/final-elf-sb-mismatch-pubkey.bin         -m 2G -nographic -vga none -nic none         -serial mon:stdio
tdx_caps->attrs_fixed0: 0x8000000060000001
tdx_caps->attrs_fixed1: 0x0
tdx_caps->xfam_fixed0: 0x6dbe7
tdx_caps->xfam_fixed1: 0x3
cpuid_config[0]: 0x1_0xffffffff eax 0x0 ebx 0xff0000 ecx 0x30044180 edx 0xb0400000
cpuid_config[1]: 0x4_0x0 eax 0xffffc3ff ebx 0xfffff000 ecx 0xffffffff edx 0xfffffffb
cpuid_config[2]: 0x4_0x1 eax 0xffffc3ff ebx 0xfffff000 ecx 0xffffffff edx 0xfffffffb
cpuid_config[3]: 0x4_0x2 eax 0xffffc3ff ebx 0xfffff000 ecx 0xffffffff edx 0xfffffffb
cpuid_config[4]: 0x4_0x3 eax 0xffffc3ff ebx 0xfffff000 ecx 0xffffffff edx 0x7
cpuid_config[5]: 0x7_0x0 eax 0x0 ebx 0xd02b9108 ecx 0x7e60 edx 0x40000
cpuid_config[6]: 0x7_0x1 eax 0x30 ebx 0x0 ecx 0x0 edx 0x0
cpuid_config[7]: 0x1c_0xffffffff eax 0xb ebx 0x0 ecx 0x0 edx 0x0
qemu-kvm: warning: host doesn't support requested feature: CPUID.0DH:ECX [bit 10]
tdx_finalize_vm: finalize vm
tdx_finalize_vm 364
error: kvm run failed Input/output error
EAX=00000000 EBX=00000000 ECX=00000000 EDX=000a06d0
ESI=00000000 EDI=00000000 EBP=00000000 ESP=00000000
EIP=0000fff0 EFL=00000002 [-------] CPL=0 II=0 A20=1 SMM=0 HLT=0
ES =0000 00000000 0000ffff 00009300
CS =f000 ffff0000 0000ffff 00009b00
SS =0000 00000000 0000ffff 00009300
DS =0000 00000000 0000ffff 00009300
FS =0000 00000000 0000ffff 00009300
GS =0000 00000000 0000ffff 00009300
LDT=0000 00000000 0000ffff 00008200
TR =0000 00000000 0000ffff 00008b00
GDT=     00000000 0000ffff
IDT=     00000000 0000ffff
CR0=60000010 CR2=00000000 CR3=00000000 CR4=00000000
DR0=0000000000000000 DR1=0000000000000000 DR2=0000000000000000 DR3=0000000000000000
DR6=00000000ffff0ff0 DR7=0000000000000400
EFER=0000000000000000
Code=00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 <00> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ??
KVM_GET_CLOCK failed: Input/output error
Aborted (core dumped)

liuw1, 2023-03-23:
Test Steps
1. bash sh_script/build_final.sh elf_sb_test
2. bash sh_script/integration_tdx.sh -t elf -f target/release/final-elf-sb-mismatch-pubkey.bin
Environment:
tdx-seam-module <2.0.00.20-5>
kernel <4.6-15.el8>
qemu-kvm <7.0.50-7.gnr_bkc_pc_1.7.el8>

gaojiaqi7, 2023-03-23:
Calling another `panic!` in panic_handler may be wrong
https://github.com/confidential-containers/td-shim/blob/main/td-shim/src/bin/td-shim/main.rs#L60

closed 2023-03-31 by jyao1