confidential-containers/td-shim #554: How to install TD-Shim

dimakuv, 2023-03-31:
I am able to run things like `cargo image -t executable -p <my payload> --release` in the latest TD-Shim, from the TD-Shim directory.

But how am I supposed to install TD-Shim on my system (say, Ubuntu 22.04) and be able to invoke the same command (or a similar command) from any directory?

I see that there is `make install` target in the Makefile, but I don't know how to properly run the Makefile... Whatever I try, I get errors:
```sh
$ make all
bash sh_script/preparation.sh
~/tmp/td-shim/library/ring ~/tmp/td-shim
fatal: Could not parse object '9cc0d45f4d8521f467bb3a621e74b1535e118188'.
can't find file to patch at input line 5
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/Cargo.toml b/Cargo.toml
|index c9daac82e..f9e578e39 100644
|--- a/Cargo.toml
|+++ b/Cargo.toml
--------------------------
File to patch:

# no idea what to do now, so I terminate this process
```

```sh
$ make build
cargo +nightly-2022-11-15 xbuild --target x86_64-unknown-none -p td-shim --release
error: cargo metadata invocation failed: Error during execution of `cargo metadata`: error: failed to load manifest for workspace member `/home/sdp/tmp/td-shim/cc-measurement`

Caused by:
  failed to load manifest for dependency `ring`

Caused by:
  failed to read `/home/sdp/tmp/td-shim/library/ring/Cargo.toml`

Caused by:
  No such file or directory (os error 2)

make: *** [Makefile:93: none-build-td-shim] Error 1
```

```sh
$ make install
cargo +1.66.0 build -p td-shim-tools --release
error: toolchain '1.66.0-x86_64-unknown-linux-gnu' is not installed
make: *** [Makefile:116: build-td-shim-tools] Error 
```

Does TD-Shim support installation currently? If yes, what are the instructions?

gaojiaqi7, 2023-04-04:
hello, do you mean installing the td-shim binary? I don't think we support it now. The `install` target installs the `td-shim-tools` and the `install-devtools` target installs the tools under `devtools`.

For the errors of `make all` and `make build`, have you initialized the submodule? From the error message, it seems that `ring` can't be found.

For the error of `make install`, please install the stable toolchain:
```
rustup install 1.66.0-x86_64-unknown-linux-gnu
```

dimakuv, 2023-04-04:
> hello, do you mean installing the td-shim binary? I don't think we support it now. The `install` target installs the `td-shim-tools` and the `install-devtools` target installs the tools under `devtools`.

I think that `td-shim-tools` without `td-shim` and `ResetVector.bin` is a bit useless? My purpose is to be able to combine my payload with `td-shim` + `ResetVector`, to get the final binary to be supplied to e.g. QEMU via `-bios` option.

> For the errors of `make all` and `make build`, have you initialized the submodule? From the error message, it seems that `ring` can't be found.

I see.

> For the error of `make install`, please install the stable toolchain:
> 
> ```
> rustup install 1.66.0-x86_64-unknown-linux-gnu
> ```

But why? Why would TD-Shim require two Rust toolchains (one stable, one nightly)? How does it even work?

gaojiaqi7, 2023-04-04:
The nightly is for crates like `td-shim` that requires unstable features, and the stable one is for tools. And of course the tools can be built with the nightly toolchain...

dimakuv, 2023-04-04:
I still don't get it. What is the use of tools (`td-shim-tools`) without `td-shim` itself? Who would want to build the tools (via stable Rust toolchain) and not the `td-shim` binary (via nightly Rust toolchain)? What am I missing?