confidential-containers/td-shim #555: Can't reproduce binary using different Linux users

gaojiaqi7, 2023-04-04:


liuw1, 2023-04-10:
I will take this

liuw1, 2023-04-13:
After investigation, the relocation section and symbol map of the output binaries generated by the Rust compiler may differ between different Linux users due to differences in file permissions and environment variables. See blow image:
[image]

Rust Reproducible Build Ref link: https://users.rust-lang.org/t/testing-out-reproducible-builds/9758

liuw1, 2023-04-13:
Recommendation:
1. Use same user name 'tdshim'
2. Use same tool configuration(same version):
    clang
    nasm
    rustc
    cargo
    cargo-xbuild
    rust-toolchain

liuw1, 2023-12-04:
It has been fixed with 2300bb85828edc37ef8bd6c2eab9880286331989.