confidential-containers/td-shim #565: Github doesn't recognize license

dcmiddle, 2023-05-25:
https://github.com/confidential-containers/td-shim/pull/252 attempted to resolve this, but there may be 2 issues.

First the recognized file names are all caps with  txt, rtf, or md extensions 
See https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/licensing-a-repository#determining-the-location-of-your-license

Second the license is 2 clause BSD with a patent grant. https://spdx.org/licenses/BSD-2-Clause-Patent
However, while SPDX recognizes the liences, Github does not have a string match for that variant
See https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/licensing-a-repository#searching-github-by-license-type

jyao1, 2023-06-11:
So, what is required action for this project?

jyao1, 2023-09-14:
@dcmiddle , we passed all license check. See front page.

Can we close it now?

closed 2023-09-14 by dcmiddle