confidential-containers/td-shim #574: Error when generating out_bin in td-shim-tee-info-hash

Xynnn007, 2023-06-13:
I got the following error when using `td-shim-tee-info-hash` to generate a MRTD value from a built `final-boot-kernel.bin`
```
[root@localhost td-shim]# target/debug/td-shim-tee-info-hash --manifest td-shim-tools/src/bin/td-shim-tee-info-hash/sample_manifest.json --image target/release/final-boot-kernel.bin -o testbin
thread 'main' panicked at 'Mismatch between definition and access of `out_bin`. Could not downcast to std::path::PathBuf, need to downcast to alloc::string::String
', td-shim-tools/src/bin/td-shim-tee-info-hash/main.rs:67:36
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Aborted (core dumped)
```

gaojiaqi7, 2023-06-19:
Hi @Xynnn007, sorry for the late response. Could you please try with the latest code? I think this issue has been fixed by https://github.com/confidential-containers/td-shim/commit/1caeba4a36cc3372ad150485611ff0742a6a254a

Xynnn007, 2023-06-26:
> Hi @Xynnn007, sorry for the late response. Could you please try with the latest code? I think this issue has been fixed by [9463ffe](https://github.com/confidential-containers/td-shim/commit/9463ffe526734e3f832793d1773e75d9c41cda7e)

It works. Thanks!

closed 2023-06-26 by Xynnn007