confidential-containers/td-shim #575: TD-Shim spec: description of the initial page table mapping?

dimakuv, 2023-06-22:
I'm using TD-Shim with my own ELF payload.

Looking at the code and experimenting a bit, I notice that the initial page tables (which are installed before the control is passed to the payload) use 2MB pages, occupy 128KB at the end of RAM but before 2GB, and map `[0..4GB)` address space.

1. Is my understanding correct?
2. Is this subject to change, or we can rely on this TD-Shim behavior in the future?
3. If this will not change, shouldn't it be part of the TD-Shim spec?

jyao1, 2023-07-31:
The general rule is: 
1) The reserved memory will be reported in E820 table.
2) The transition state will be in reserved memory.
3) Once it is transferred, the payload shall setup its own state. (page table, GDT, IDT, stack, heap, etc)

It is implementation choice that the reserved memory is on the top of RAM below 4GB, which aligned with real platform BIOS.
I don't recommend you rely on the behavior. Please follow the general rule.

dimakuv, 2023-07-31:
@jyao1 But the E820 table shows only the guest physical memory? How can I deduce the installed page tables from the E820 table (i.e., what guest virtual memory is accessible at the executable start)?

jyao1, 2023-07-31:
@dimakuv, I am not sure if I understand your question.

E820 is a common way to describe the physical memory. The detail is described in ACPI specification.
Why you cannot do it?

dimakuv, 2023-07-31:
In my executable entry point, I have a tiny x86-64 assembly loader. This asm loader simply installs a stack and jumps into the C startup code. Then at some point, the C startup code will set up its own page tables, but before that, the C code simply assumes that there are some initial page tables (set up by TD-Shim).

So far, this arrangement worked good for my case, but from your answers here, it looks like I need to modify my asm loader: it needs to set up the page tables immediately, before passing control to my other code (that has some assumptions on what virtual memory can be accessed and with which permissions).

jyao1, 2023-07-31:
Not only **page table**, but also **GDT**, **IDT**. All X86 arch state.

The memory access rule is defined in ACPI E820. The payload can use "usable" memory, but it cannot touch "reserved" memory.
Please see https://uefi.org/specs/ACPI/6.5/15_System_Address_Map_Interfaces.html.

dimakuv, 2023-07-31:
> Not only page table, but also GDT, IDT. All X86 arch state.

Ok, I get it now. So my loader (in x86-64 assembly) must set up my own page tables first. It must not rely on the TD-Shim installed page tables. Thanks.

closed 2023-07-31 by dimakuv