confidential-containers/td-shim #577: [RFC] td-shim release with reference MRTD

Xynnn007, 2023-06-26:


jyao1, 2023-07-31:
@Xynnn007 , please submit issue and PR.

Xynnn007, 2023-10-08:
I have made a test repo to build `td-shim` and reference values here https://github.com/Xynnn007/rim-generator/releases/tag/v0.1.0 

IMO CoCo will have multiple components together with their reference value to be built. Let's take TDX for example:
- td-shim: `final-boot-kernel.bin`
- guest-kernel: `bzImage`
- guest-image: `kata-containers.img`

A first impression is that we use `td-shim` repo to publish td-shim and its reference value, and we use `kata-containers` repo to publish guest image and its reference value. However, putting the release processes of these components in different repositories will be unfriendly to users and also hard for community to maintain. Is it a good choice to use a separate repository to maintain the publishment related scripts and artifacts? like https://github.com/confidential-containers/infra or https://github.com/confidential-containers/confidential-containers or a new repo?

@sameo @fitzthum @jyao1 @peterzcst @dcmiddle @mythi @arronwy @jialez0

jyao1, 2023-10-12:
I am OK if it aligns to other project.

fitzthum, 2023-10-13:
I think you're on the right track. My biggest question is how we can integrate reference artifacts with our existing build pipelines. For example for td-shim we use versions.yaml file to track which version will be built into the payload container. Should this also be where we generate reference measurements or should we try to decouple the building of our artifacts from the Kata CI?

Also, are we making some assumptions about how different platforms will measure components? For `td-shim` I guess we only have to worry about TDX (for now, at least), but for other components, like `OVMF` I'm a little fuzzy on how we can provide a reference measurement that will be compatible with different platforms or even different hypervisors.

jyao1, 2023-10-16:
OVMF can support one binary for TDX/SEV/non-CC.

Xynnn007, 2023-10-16:
> I think you're on the right track. My biggest question is how we can integrate reference artifacts with our existing build pipelines. For example for td-shim we use versions.yaml file to track which [...]

I think we get the same question. There are two ways I found
1. We integrate the `reference value calculation` part to current Kata CI.
2. We create a separate repo to maintain all the artifact releases, related GHA and scripts.

As we are planning to make a website for CoCo, it would be great to have a download page that points to the hyperlink of the artifacts together with their reference values.

I'd also like to hear some point views of @fidencio for this.

> Also, are we making some assumptions about how different platforms will measure components? For td-shim I guess we only have to worry about TDX (for now, at least), but for other components, like [...]

A reference value does not necessarily correspond to one component, it may correspond to multiple components, which depends on how TEE measures. For example, TDX measures guest firmware like td-shim/OVMF and store the measurements into MRTD register. Thus in RVPS we use key `tdx.quote.body.mr_td` to look up the reference value. However, SNP measures OVMF/kernel/kernel cmdline/guest rootfs togetherly and store the measurement into one [register](https://docs.rs/sev/1.2.1/sev/firmware/guest/struct.AttestationReport.html) `measurement`. Thus in RVPS we use key `snp.measurement` to look up the reference value for a combination of OVMF/kernel/kernel cmdline/guest rootfs in RVPS.

This hints that we should release a binary bundle for different platforms together with the reference values. Back to the question itself. If we release a same OVMF binary for both TDX/SEV, we would have two reference value files for each platforms. 

Other platforms should have their different mechanisms but we can fit well with the design.

fitzthum, 2023-10-16:
> I think we get the same question. There are two ways

Some components, like OVMF, seem like they could be decoupled from the Kata build system fairly easily. To me it seems a bit weird that Kata is building OVMF. Other components, like the rootfs, are more closely tied to Kata and would be tricky to replicate elsewhere. I suspect that Kata is somewhat attached to building components themselves, which does enable some significant tooling such as the kernel patching/building scripts.

On the other hand, we also have the image buiders for peer pods, which already duplicate some of the functionality of the kata tooling. That's another place we'll need to come up with some reference values.

> This hints that we should release a binary bundle for different platforms together with the reference values. 

Yeah, I agree. Reference values will have to be platform specific.

Btw this is something that [igvm](https://github.com/microsoft/igvm) might be able to help out with.

jyao1, 2023-10-19:
> Btw this is something that [igvm](https://github.com/microsoft/igvm) might be able to help out with.

I filed https://github.com/confidential-containers/td-shim/issues/610 for IGVM.
Does KVM/QEMU support IGVM format? or Hyper-V only?

fitzthum, 2023-10-24:
> Does KVM/QEMU support IGVM format? or Hyper-V only?

I don't think there is QEMU support yet, but I've heard a rumor that someone is working on it.

Hyper-V as well as CLH seem to be supported. There is some info [here](https://github.com/microsoft/igvm-tooling#cloud-hypervisor-support).

fitzthum, 2023-11-01:
Btw another thing about the current release process is that the tagging of releases, which is typically done whenever we reach that step in the release checklist, is not really in sync with development. We just come through and tag the repo whenever the CoCo project happens to be doing a release. So far this doesn't seem to have caused any issues, but it doesn't seem ideal.