confidential-containers/td-shim #595: CI: fuzz may fail

gaojiaqi7, 2023-09-12:
CI output:

Test Case: td-uefi-pi - afl_payload_parser fail
~/work/td-shim/td-shim
~/work/td-shim/td-shim/td-shim ~/work/td-shim/td-shim
Test Case: td-shim - afl_secure_boot_cfv fail
Test Case: td-shim - afl_secure_boot_payload fail

liuw1, 2023-09-13:
I will take this task

jyao1, 2023-09-14:
if the failure is caused by "no enough CPU core", then can we reduce the required CPU core number?

liuw1, 2023-09-14:
For fuzzing CI, we only need 1 cpu core in same period. 

So it's random issue, we can not predict when the cpus will be locked by other tasks. I tried to use command `afl-gotcpu` to check cpu status before run AFL test case. But it doesn't work.

I will keep to find solutions.

PR #598 (2023-09-14): Fix random fail issue of CI fuzzing test
Fix #595
closed 2023-09-14 by jyao1