confidential-containers/td-shim #599: Consider add `Cargo.lock`

gaojiaqi7, 2023-09-15:
As this project poduces binaries and tools, it is reasonable to add a `Cargo.lock` file to provide deterministic builds all the time [1].

In the current state, we may encounter many problems (#583 #491 #354 etc.) caused by unfixed dependent versions that break the build, such as changes in dependencies' APIs and required rust versions.

On the one hand, previous releases may not compile correctly at any time in the future then all the projects that depends on td-shim will break, so we must make emergency updates to fix these issues. On the other hand, automatic update of dependencies may also bring potential runtime issues.

I suggest to add a `Cargo.lock` file in the repository and use github dependabot [2] to automatically upgrade the version of dependencies in a fixed time interval. Community repositories (attestation-service)[https://github.com/confidential-containers/attestation-service] (guest-components)[https://github.com/confidential-containers/guest-components] and (kbs)[https://github.com/confidential-containers/kbs] both use this kind of flow.

[1] https://doc.rust-lang.org/cargo/faq.html#why-do-binaries-have-cargolock-in-version-control-but-not-libraries
[2] https://docs.github.com/en/code-security/dependabot/dependabot-version-updates

gaojiaqi7, 2023-09-15:
@jyao1 any comments?

jyao1, 2023-09-15:
Agree to follow what the rest components do.

closed 2023-09-18 by jyao1