confidential-containers/td-shim #603: What are the Payload Parameters for an custom ELF payload?

dimakuv, 2023-09-21:
The TD-Shim spec has this:

<html>
<body>

Boot Configuration | RTMR[1] | YES | TD-Shim: TDCALL[TDG.MR.RTMR.EXTEND] | Payload parameter - command line | 5
-- | -- | -- | -- | -- | --

</body>
</html>

See https://github.com/confidential-containers/td-shim/blob/main/doc/tdshim_spec.md#td-measurement

I can understand what is included in the payload parameter in case of the Linux kernel.

But if I have my custom ELF payload, what constitutes the payload parameters exactly? Looks like at least these are included:
- E820 table 
- all ACPI tables

Is there anything else? What is the exact code that extends RTMR[1] with these parameters?

Also, do I understand correctly that for custom ELF payloads, the "command line" doesn't make any sense and can be ignored?

jyao1, 2023-10-05:
I dont think E820 table and ACPI are RMTR[1].

The rule is: VMM input must be in RTMR. VMM input for TDShim in RTMR[0] as firmware config, VMM input for Payload is RTMR[1] as Payload/OS config.

E820 and ACPI are internal data generated by TdShim. There is no need to measurement them into RTMR[1].

jyao1, 2023-10-05:
If your payload does not have any "command line", then you can just ignore it. No problem at all.

dimakuv, 2023-10-09:
Thanks, this answers my question.

closed 2023-10-09 by dimakuv